sipeed/MaixPy-v1 #8: Compile instructions error

loboris, 2018-11-16:
I think there is a small error in in compile instructions in **README.md**.

The **_Kendryte compiler path_** should be **added** to the `$PATH` & `$LD_LIBRARY_PATH`, like this
```
export PATH=$PATH:<path_to_kendryte-toolchain/bin>
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:<path_to_kendryte-toolchain/bin>
```
Than make can be run with:
```
make CROSS_COMPILE=riscv64-unknown-elf-
```
Or the `CROSS_COMPILE`  can also be exported: `export CROSS_COMPILE=riscv64-unknown-elf-`

At least, for me it works only this way. A simple bash file can be created for compiling:
```
#!/bin/bash
export PATH=$PATH:<path_to_kendryte-toolchain/bin>
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:<path_to_kendryte-toolchain/bin>
export CROSS_COMPILE=riscv64-unknown-elf-
make

```

xiaoxiaohuixxh, 2018-11-22:
yes,i will fix it,ths a lot!

wipping, 2018-11-30:
We have fixed it , you can have a try .

closed 2018-11-30 by wipping

loboris, 2018-11-30:
Thanks, it works now.