sipeed/MaixPy-v1 #24: Pre-built firmware on GitHub

DurandA, 2019-01-05:
It would be nice to have pre-built images on the [relases page](https://github.com/sipeed/MaixPy/releases).

I am aware of the [firmware download link on bbs.lichee.pro](http://bbs.lichee.pro/d/170-maix-py-lichee-dan-sipeed-m1-mpy-bin) but this painful to use for non-chinese users.

xiaoxiaohuixxh, 2019-01-15:
we will post the images before January 23.

Novakov, 2019-01-31:
@xiaoxiaohuixxh Any update on this? I just received my Maix GO board and after uploading firmware.bin it's not booting - 'boot failed with exir code 233'

Neutree, 2019-03-02:
of course we do, 

@Novakov  download method see [doc](https://maixpy.sipeed.com/en/get_started/upgrade_firmware.html), if you still have problem, create a new issue please

closed 2019-03-02 by Neutree