sipeed/MaixPy-v1 #25: Cannot use rshell or ampy

DurandA, 2019-01-05:
This version of MicroPython is somehow not compatible with *rshell* nor *ampy*. These are tools for interacting with MicroPython's filesystem.

This might be related to non-standard debug output that confuse these tools.

If this issue should not be resolved, what is the most practical way to upload scripts to MaixPy?

wipping, 2019-01-15:
The current release of MaixPy does not support most of these features, we are developing another version of MaixPy, this version of MaixPy will support the fat file system. In addition, zmodem can be used to transfer small size script files to the development board.

goatchurchprime, 2019-01-17:
It worked on my jupyter_micropython_kernel first time!

https://github.com/goatchurchprime/jupyter_micropython_kernel/

The code I tried it out on is here:  https://github.com/goatchurchprime/jupyter_micropython_kernel/blob/master/Maixpy_demo.ipynb

robert-hh, 2019-01-22:
Ampy get fails, because file read on MaixPy returns '' at the end of the file, and the code ampy uploads expects b'' as the EOF signal. Funny enough, it works with an ESP32 with also returns ''.
Workaround: change ampy, files.py, line 64 into `if not result:`? At least for me that worked.

robert-hh, 2019-01-28:
For ampy I have made a PR that fixes it (https://github.com/sipeed/MaixPy/pull/33). For rshell more changes are required, and I do not know if it is worth the effort. It works now for me. The changes are:
- board.name shall return a board name. That can be done by assigning one in boot.py.
- os.stat("/") shall return a tuple with 0x4000 in the first element. Since nitrogen spiffs does not know directories, that would require a special treatment of that name, and I doubt that it is worth the effort.

Neutree, 2019-03-02:
we support ampy and GUI tool uPyloader now, but not support rshell yet, see [doc](https://maixpy.sipeed.com/en/get_started/upload_script.html)

we will close this issue, and if have question you can reopen or create a new one, thanks

closed 2019-03-02 by Neutree