sipeed/MaixPy-v1 #26: Undefined reference to `pthread_setcancelstate'

Erhaede, 2019-01-14:
During the compilation of the port, I get this error  and I have no idear of how to solve it. I have used kendryte-gnu-toolchain and the official kendryte-toolchain-8.2 and I get the same error. 

Do you have any idear of where this is comming from? 

`LINK micropython
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-vfiprintf.o): in function `.L13':
vfprintf.c:(.text._vfiprintf_r+0x32): undefined reference to `pthread_setcancelstate'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-vfiprintf.o): in function `.L15':
vfprintf.c:(.text._vfiprintf_r+0x76): undefined reference to `pthread_setcancelstate'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-vfiprintf.o): in function `.L16':
vfprintf.c:(.text._vfiprintf_r+0x9a): undefined reference to `pthread_setcancelstate'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-vfiprintf.o): in function `.L0 ':
vfprintf.c:(.text._vfiprintf_r+0x858): undefined reference to `pthread_setcancelstate'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fflush.o): in function `.L47':
fflush.c:(.text._fflush_r+0x24): undefined reference to `pthread_setcancelstate'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/lib/libc.a(lib_a-fflush.o):fflush.c:(.text._fflush_r+0x3a): more undefined references to `pthread_setcancelstate' follow
collect2: error: ld returned 1 exit status
`

xiaoxiaohuixxh, 2019-01-15:
What are the systems and branches you use?

Erhaede, 2019-01-15:
I'm using Linux  4.4.0-34-generic #53-Ubuntu x86_64 GNU/Linux and the master branch.

Erhaede, 2019-01-19:
SOLVED: 

Add 		$(SDK_PATH)/lib/bsp/pthread.c 
to SRC_C = \   in the Makefile like this: 

SRC_C = \
		$(MODULES_SRC_C)\
		main.c \
		mphalport.c \
		modutime.c \
		$(SDK_PATH)/lib/bsp/entry.c \
		$(SDK_PATH)/lib/bsp/pthread.c 
                ... 

and copy the "pthread.c" file from the SDK 0.7.0 to /MaixPy-master/ports/k210-standalone/kendryte-standalone-sdk/lib/bsp/pthread.c

Hope it helped ;)

closed 2019-02-25 by Erhaede