sipeed/MaixPy-v1 #27: Missing library 'time'

goatchurchprime, 2019-01-17:
'import time' doesn't work.  

This is a very important core module and makes me worried about what else is missing!  

You can't do 'help("modules")' to get a list of what is there.  

This is a nice board to use, but it is a problem if it does not have the functions I used to use.

pfalcon, 2019-01-18:
@goatchurchprime: You should know that MicroPython doesn't have "time" module, only "utime": http://docs.micropython.org/en/latest/library/utime.html

If you need "time", you'll either need to make your own, or find somebody else's who did that.

time.py:
~~~
from utime import *

# More of your stuff here
~~~

robert-hh, 2019-01-19:
utime also is not implemented. And the micropython base port defines time as the alias to utime.

frdnd, 2019-01-21:
It would be nice to include utime.ticks_us(), utime.ticks_diff() into the micropython port

robert-hh, 2019-01-22:
In my box of patches I have files that add the ticks_xxx set of methods as a fast & incomplete fix
https://github.com/robert-hh/K210-Stuff
It consist of the files modutime.c, mphalport.c and mphalport.h. You have to build you own image to get that working. It is incomplete since at the moment it assumes a CPU clock of 160 MHz. To be improved later.

Neutree, 2019-03-02:
module time exist in latest code and release

closed 2019-03-02 by Neutree