sipeed/MaixPy-v1 #31: Trial image with some fixes

robert-hh, 2019-01-24:
First of all: This is no bug report, just a hint!
In the past days I have commented to quite a few issues, and also made some patches to these issues. The changed code parts are available here: https://github.com/robert-hh/K210-Stuff
For those who do not like to build their own image, I also uploaded the latest image with these changes.
Flashing can be done following these instructions:
"In Windows, download K-Flash from https://kendryte.com/downloads/ ,  unzip and run K-Flash.exe
In K-Flash select the device’s port"
In Linux, use the command kflash.py from https://github.com/kendryte/kflash.py and https://github.com/kendryte/kendryte-flash-windows/blob/develop/src/Canaan.Kendryte.Flash/Resources/isp_flash.bin
' ./kflash.py -p /dev/ttyUSB0 -b 2000000 -l isp_flash.bin sipeedm1.kfpkg'

frdnd, 2019-01-26:
Thx for all the work :)

robert-hh, 2019-01-26:
Your welcome. I am creating now a series of PR's to introduce the changes piece by piece.

namn, 2019-02-13:
So I flashed your image in. How do I use it? When I do `screen /dev/ttyACM0 115200`, the device turns black, and `screen` also shows a black screen. Nothing seems to be running. Sorry for the noobie question.

robert-hh, 2019-02-13:
The port to use did not change. On linux, it is /dev/ttyUSB0 ...

namn, 2019-02-13:
I was able to flash with `/dev/ttyACM0` though. And my system does not have `/dev/ttyUSB0`. When I removed the device, `/dev/ttyACM0` disappeared. When I plugged it back in, `/dev/ttyACM0` appeared again. I'm using the included A-to-C USB cable to plug the device to my desktop (C head connected to the Maix GO, A head connected to my desktop).

robert-hh, 2019-02-13:
So you should be right. I have quite q few different modules here, and gave away my MaixPy device to  a friend at the moment. After you started it with screen, you should have a Python REPL prompt. Be sure to disable any flow control on the screen program, option -fn. I actually use picocom as terminal emulator. very small and simple.

robert-hh, 2019-02-20:
Since sipeed published a new version which includes many fixes, this image is kind of obsolete.

closed 2019-02-20 by robert-hh