sipeed/MaixPy-v1 #32: file.read() returns the wrong data type for files opened as binary

robert-hh, 2019-01-25:
When a file is opened in binary mode, like f = open("name", "rb"), a file.read() must return a bytes object. Instead, the MaiyPy implementation returns a string object,

pfalcon, 2019-01-25:
> When a file is opened in binary mode, like f = open("name", "rb"), a file.read() must return a bytes object. Instead, the MaiyPy implementation returns a string object,

What's funny is that it's totally impossible in any properly made port of MicroPython. MicroPython implementation itself provides implementation of all file functions (.read(), .write(), .seek(), .flush(), etc.) which work correctly regarding return types, etc..

The porter just need to supply low-level implementations of POSIX-like functions for the underlying file system.

robert-hh, 2019-01-25:
Aggreed. Thanks Paul. I looked into the code and did not spot yet in the low level implementation at file_io.c where a difference is being made. But the code is still in an early stage, and a lot is missing. I am even not sure whether the code in the repository is up-to-date.

robert-hh, 2019-01-25:
I think I found the cause. In file_io.c , fileio_stream_p is deinfed as:
```
STATIC const mp_stream_p_t fileio_stream_p = {
    .read = file_obj_read,
    .write = file_obj_write,
    .ioctl = file_obj_ioctl,
};
```
The structure element is:_text is not defined. And since this defaults to true, the "binary" steram is handles as text stream. I'll check that this evening.

robert-hh, 2019-01-25:
Found the reason in mpconfigport.h as:
#define MICROPY_PY_IO_FILEIO                (0)
should be:
#define MICROPY_PY_IO_FILEIO                (1)

robert-hh, 2019-01-25:
Change to configboard.h:
1 line changed,
```
#define MICROPY_PY_IO_FILEIO (1)
```
 2 lines added:
```
#define mp_type_fileio                      mp_type_vfs_spiffs_fileio
#define mp_type_textio                      mp_type_vfs_spiffs_textio
```

Neutree, 2019-03-02:
fixed in latest code, thanks very much

closed 2019-03-02 by Neutree