sipeed/MaixPy-v1 #37: Illegal instruction when connecting to network

jeremytregunna, 2019-02-25:
When I try and connect to a wireless network, I get an illegal instruction, which crashes the K210 dumping registers and a hard reset is required.

This is using the 0.1.1 prerelease from github. Code below illustrates the problem:

```
>>> import network
>>> network.ESP8285.connect("ssid", "password")
core dump: illegal instruction
Cause 0x0000000000000002, EPC 0x0000000000000000
reg[00](zero ) = 0x0000000000000004, reg[01](ra   ) = 0x000000008007a138
reg[02](sp   ) = 0x00000000804c43b0, reg[03](gp   ) = 0x000000008014f068
reg[04](tp   ) = 0x00000000804bcbc0, reg[05](t0   ) = 0x0000000000000002
reg[06](t1   ) = 0x0000000080013c0c, reg[07](t2   ) = 0x0000000000000000
reg[08](s0/fp) = 0x0000000000000010, reg[09](s1   ) = 0x0000000000000000
reg[10](a0   ) = 0x0000000000000000, reg[11](a1   ) = 0x0000000080146a20
reg[12](a2   ) = 0x000000000000000a, reg[13](a3   ) = 0x00000000804c45bc
reg[14](a4   ) = 0x0000000000000000, reg[15](a5   ) = 0x0000000000000000
reg[16](a6   ) = 0x00000000800acf40, reg[17](a7   ) = 0x0000000000000002
reg[18](s2   ) = 0x0000000000000000, reg[19](s3   ) = 0x0000000000000000
reg[20](s4   ) = 0x00000000000000c0, reg[21](s5   ) = 0x0000000000000001
reg[22](s6   ) = 0x00000000000000b0, reg[23](s7   ) = 0x00000000800a88f0
reg[24](s8   ) = 0x0000000080158901, reg[25](s9   ) = 0x0000000000000000
reg[26](s10  ) = 0x0000000000000000, reg[27](s11  ) = 0x0000000000000000
reg[28](t3   ) = 0x0000000000000003, reg[29](t4   ) = 0x0000000000000002
reg[30](t5   ) = 0x0000000000000000, reg[31](t6   ) = 0x000000000000022a
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149420()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149430()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149440()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149450()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149478()
freg[10](fa0 ) = 0x4000000000000000(), freg[00]() = 0x00000000801494a8()
freg[12](fa2 ) = 0x41d8054ac0000000(), freg[1610612736]() = 0x00000000801494d0()
freg[14](fa4 ) = 0x41d8054ac0000000(), freg[1610612736]() = 0x00000000801494e0()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x00000000801494f0()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149518()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149528()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149538()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149548()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x0000000080149558()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x0000000080149580()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x0000000080149590()
W (9313734513) SYSCALL: sys_exit called by core 0 with 0x539
```

wipping, 2019-02-27:
Your code is wrong.
For this question, you can check out the MaxiPy wiki,(which currently only supports Chinese).You can try to check it out.
wiki addr:https://maixpy.sipeed.com/

jeremytregunna, 2019-02-27:
The example at the bottom does show a working solution, thanks.

closed 2019-02-27 by jeremytregunna