sipeed/MaixPy-v1 #38: 有关board_info的问题

Ray0117, 2019-02-27:
逐条语句进行输入的时候board_info.可以正常使用，但是在文件中使用就会出现NameError: name 'board_info' isn't defined，代码参考PWM的例程1

Neutree, 2019-02-27:
这个类是在 `_boot.py`文件里面导入的，可以自己导入,类似的 `fm`对象也是
```python
from board import board_info
from fpioa_manager import *
```

Ray0117, 2019-02-27:
谢谢！我还遇到一个问题就是运行完一个py文件后再运行其他的就没有输出和反应了，不知道是和什么原因有关。
[image]

Ray0117, 2019-02-27:
谢谢，导入board_info解决了一部分的问题 ，但是还是会提示`AttributeError: type object 'board_info' has no attribute 'LED_G'`。
另外文档中提到的运行文件方式是通过import实现的，但是这样在运行一次之后就必须要reset才能继续使用不然就没有结果，请问有没有好的解决方法。

wipping, 2019-02-27:
请参考_boot.py的用法，导入board_info的是个类，需要开辟对象才能用
```
from board import board_info
board_test = board()
board_test.LED_G
```

closed 2019-02-28 by Ray0117