sipeed/MaixPy-v1 #41: Maix Go with putty on windows doesn't works

seksity, 2019-03-14:
I using putty on windows to terminal to Maix Go with baudrate 115200N81 but when open port the LCD display on Maix Go it cleared the welcome message ("Welcome to MaixPy") then i cannot type any command on putty.

while if i upgrade the firmware with K-Flash finished it will automatically open serial port terminal, that i can use it to key any command such as help().

How can i fixed it to works with putty?

robert-hh, 2019-03-14:
Did you set flow-control off in Putty?

seksity, 2019-03-14:
Yes, I did. I also test any baudrate and flow control to None it doesn’t
works

stheil, 2019-03-15:
> Yes, I did. I also test any baudrate and flow control to None it doesn’t works
> […](#)

I had the same problem but I found the correct settings for PuTTY. I will put them here when I'm back home at my PC tonight.

seksity, 2019-03-15:
Thank you, waiting you

stheil, 2019-03-15:
Sorry it took so long. It works for me with all settings to default, except for the baud rate (115200) and the flow control set to RTS/CTS. I thought it was supposed to work without flow control but apparently they use one of those lines as a reset signal.

Neutree, 2019-03-16:
try [open-ec firmware](https://github.com/sipeed/open-ec/releases) for STM32 on GO board maybe work?

seksity, 2019-03-16:
> Sorry it took so long. It works for me with all settings to default, except for the baud rate (115200) and the flow control set to RTS/CTS. I thought it was supposed to work without flow control but apparently they use one of those lines as a reset signal.

> Sorry it took so long. It works for me with all settings to default, except for the baud rate (115200) and the flow control set to RTS/CTS. I thought it was supposed to work without flow control but apparently they use one of those lines as a reset signal.

Thanks @stheil it 's works.

seksity, 2019-03-16:
> try [open-ec firmware](https://github.com/sipeed/open-ec/releases) for STM32 on GO board maybe work?

I am have not ST-Link to upgrade STM32 firmware

closed 2019-03-16 by seksity