sipeed/MaixPy-v1 #42: * [bug] ESP8285 connect/isConnected function do not work.

snowyu, 2019-03-18:
See below:

```python
# MicroPython c23571e-dirty on 2019-03-03; Sipeed_M1 with kendryte-k210
import network
from machine import UART
fm.register(board_info.WIFI_RX,fm.fpioa.UART2_TX)
fm.register(board_info.WIFI_TX,fm.fpioa.UART2_RX)
uart = UART(UART.UART2,115200,timeout=1000, read_buf_len=4096)
nic=network.ESP8285(uart)
nic.connect("your_ssid","your_password.")
# report error: OSError: could not connect to ssid=yourssid, key=your-password
# but it's already successful.
nic.ifconfig()
# ('192.168.0.200', '255.255.255.0', '192.168.0.1', '0', '0', '00:16:17:7A:0B:0A', 'your-ssid')
# the ifconfig has alread get the ip.
nic.isconnected()
# false
```

Neutree, 2019-04-03:
It works here, maybe you forget fill the SSID and Password? Or plug the antenna in ?

And indeed, the AT WiFi module(esp8285) driver need optimize

snowyu, 2019-04-04:
It's ok now when I flash to MicroPython v0.2.4.

closed 2019-04-04 by snowyu