sipeed/MaixPy-v1 #43: os.stat('/flash/boot.py') returns ENOENT

dhylands, 2019-03-19:
I'm trying to get rshell working on th maixpy and I ran into the following issue.

I built the firmware following the instructions in the README.

Pressing Control-D at the REPL reports:
```
[MaixPy]: soft reboot
heap0=8012bdb0
[MaixPy] sd_init | SD_CMD0 is FF

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|

Official Site : https://www.sipeed.com
Wiki          : https://maixpy.sipeed.com

MicroPython v0.1.1-114-g094cbf618 on 2019-03-18; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
```
Here's a transcript from the REPL:
```
>>> import os
>>> os.listdir('/flash')
['/logo.bin', '/demo_pwm.py', '/demo_cam.py', '/demo_gpio.py', '/demo_lcd.py', '/PT.py', 'boot.py', '/demo_timer.py', '/init.py']
>>> os.stat('/flash/init.py')
(32768, 0, 0, 0, 0, 0, 723, 0, 0, 0)
>>> os.stat('/flash/boot.py')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```
So the question is why does os.stat('/flash/boot.py') return ENOENT when os.listdir indicates that the file is actually there?

robert-hh, 2019-03-19:
I had that working once, while I tried with the MaixPy. As you know, rshell expects "PYB: soft reboot" after soft-reboot, so I changed that text in the MaixPy code. You could also change what rshell expects. I see above in uos.listdir() a difference on how the files are shown, '/init.py' vs. 'boot.py'.  That may cause the problem.

P.S.: I cannot test it at the moment. I have given away my MaixPy device to a friend, and re-ordered new ones, which may arrive sometime in April.

dhylands, 2019-03-19:
Good point about the leading slash. I have a MaixPy, I just hadn't had time to play with it until now. I'll be investigating other issues as well, and hopefully be able to create some PRs.

Neutree, 2019-03-21:
resolved in latest code or release

closed 2019-03-21 by Neutree