sipeed/MaixPy-v1 #44: SD card not working

lovettchris, 2019-04-02:
Your docs talk about using "import sdcard", but that module doesn't exist in your demo image from [maixpy_v0.2.4_full_with_model_at_0x300000.kfpkg](http://dl.sipeed.com/MAIX/MaixPy/release/maixpy_v0.2.4/).

I'm using [Samsung Evo Select 32gb card](https://www.amazon.com/gp/product/B06XWN9Q99/ref=ppx_yo_dt_b_asin_title_o09_s00?ie=UTF8&psc=1) which is class 10.

Am I missing a step in getting this card to be recognized?

Neutree, 2019-04-03:
No `sdcard` module there, maybe you find the wrong doc? 

list files by 
```python
import os

os.listdir("/")
os.listdir("/sd")
```

lovettchris, 2019-04-03:
I'm getting this error:
```
>>> os.listdir("/sd")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```
is there some driver code I can add somewhere to fix this?

Neutree, 2019-04-04:
that means the driver can not find the SD card,
the driver should support SDHC card (support my SDHC here) but not as it shows, so maybe some bug in the SD driver,

you can find out the driver code here:[sdcard.c](https://github.com/sipeed/MaixPy/blob/master/ports/k210-freertos/platform/drivers/sdcard.c)

maybe we should rewrite this SD card driver

lovettchris, 2019-04-04:
Thanks, I added this print statement and it is firing saying sd_get_response() returned 1... so I guess my card is not supported...
```
	if(1 == sd_version)
	{
		sd_send_cmd(SD_CMD16, 512, 0);
        result = sd_get_response();
		if (result != 0x00) 
		{
			sd_end_cmd();
            printk("[MaixPy] %s | SD_CMD16 is %X\r\n", __func__, result);
			return 0xFF;
		}	
	}
```
I also think it is handy to add this 'else' block in main.c that prints a message if the card is not present...
```
        if (sdcard_is_present()) {
            spiffs_stat  fno;
            // if there is a file in the flash called "SKIPSD", then we don't mount the SD card
	    if (!mounted_flash || SPIFFS_stat(&spiffs_user_mount_handle.fs,"SKIPSD",&fno) != SPIFFS_OK){
	          mounted_sdcard = init_sdcard_fs();
	    }
        }
        else {
            printk("sdcard is not present (or supported)\r\n");
        }
```

Neutree, 2019-04-04:
The [Arduino's SD driver](https://github.com/sipeed/Maixduino/tree/master/libraries/SD) may have better compatibility, 

so the driver of maixpy should be rewrite someday~

lovettchris, 2019-04-04:
I see, I'll take a look at that.  I can't find one of the SD cards you recommend, can you send me a link to where I can buy one of these?

[image]

Pogsquog, 2019-04-06:
I also found that the sd card did not work on my maix bit when I first tried it. 
I formatted the SD card with the official sd card formatter, reseated it, and then reset the board and power cycled it a bunch of times, and it came up OK after that... not sure which particular step it was that fixed it!
I am using a sandisk ultra 8 Gb, which is fairly common sdhc class 10 card.

As a slight aside, I loaded the model from [dl.sipeed.com](http://dl.sipeed.com/MAIX/MaixPy/model/) myself using kflash.py, and the location where it ended up was 0xd00000, not 0x300000, on my device at least.

robert-hh, 2019-04-07:
I have seen a similar effect, in that it might require a few reset attempts until the SD card is mounted. But then it works fine, different to the Spiffs file system in flash which is buggy.

kartur, 2019-04-23:
> I have seen a similar effect, in that it might require a few reset attempts until the SD card is mounted. But then it works fine, different to the Spiffs file system in flash which is buggy.

I can confirm this. Usually one reset fixes it.

Also for me my SanDisk (Ultra 32GiB) seems to work best. Toshiba (4 GiB) works mostly and Samsung (Evo Plus 32GiB) didn't work at all

lovettchris, 2019-05-10:
Thanks, I got myself a SanDisk Ultra 32gb, and it works great.  Happy camper.  Interestingly running the demos off a fast SD card also increases the frame rate of the video.  Perhaps the python interpreter is pinging the slow flash memory occasionally during code execution.

closed 2019-05-10 by lovettchris

masterhapero, 2020-12-14:
FYI. There seems to be a page for cyberEye on SD compatibility matrix.

https://www.yuque.com/tinkergen-help-en/mark/failure_analysis

[image]

diazGT94, 2021-02-08:
I'm using a SanDisk 16gb class 10 HC A1 in FAT32. It happens that after some time it doesn't recognize the SD and I need to reset the SD and burn the Firmware to the board again, which is not suitable for my application. Is there any way to fix this problem?

THuffam, 2022-05-22:
Having a similar issue - SD not mounting on maixpy card with maixpy_v0.6.2_78_g469c985e6_minimum_with_kmodel_v4_support.bin but the same card works on the default firmware (v0.5.0).
Is that because the minimum firmware does not support SD card? if so thats crazy for ML work where I need to store my kmodel file.