sipeed/MaixPy-v1 #46: Face detector demo not working

lovettchris, 2019-04-02:
the [face detector demo](http://dl.sipeed.com/MAIX/MaixPy/model/face.kfpkg) from [this page](https://maixpy.sipeed.com/zh/libs/Maix/kpu.html) is not working on Maix Go.  
```
>>> task=kpu.load(0xd00000)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: [MAIXPY]kpu: load error -2
```
but the yolo demo works great, so I know my devkit is running ok.

Neutree, 2019-04-03:
The address wrong( `0xd00000`)

The firmware `maixpy_v0.2.4_full_with_model_at_0x300000.kfpkg` burn model at `0x300000`, so you load successfully as you say in #45  with `kpu.load(0x300000)`,  

This model package:[face.kfpkg](http://dl.sipeed.com/MAIX/MaixPy/model/face.kfpkg) default burn into `0xd00000`, you can edit address by decompress the kfpkg (zip format), so if you want to `kpu.load(0xd00000)`, you can burn this model by command `python3 kflash.py -p /dev/ttyUSB1 -b 2000000 -B goE face.kfpkg`

And the model in `maixpy_v0.2.4_full_with_model_at_0x300000.kfpkg`  and [face.kfpkg](http://dl.sipeed.com/MAIX/MaixPy/model/face.kfpkg) are the same, just the address different when burn to flash

closed 2019-04-04 by lovettchris

keithr0, 2019-05-13:
I am having problems running this example on Maix Go, the error is in this line

task = kpu.load(0x300000)

v=596458738, flag=-2096411809, arch=-1955282835, layer len=834782759, mem=-392490997, out cnt=-663129117
err: we only support V3 now, get V596458738
[MAIXPY]kpu: kpu_model_get_size  error -3

ValueError: [MAIXPY]:kpu load error: read model size error 1
MicroPython v0.3.1 on 2019-04-27; Sipeed_M1 with kendryte-k210