sipeed/MaixPy-v1 #47: How to flip the video?

lovettchris, 2019-04-04:
When I mount the camera on the back the video on the screen seems reversed.  When I pan right the video on screen moves left.  Is there a way to flip the video at the driver level to correct for this?

lovettchris, 2019-04-04:
Never mind, I found this method on the sensor:
```python
sensor.set_hmirror(1)
```

lovettchris, 2019-04-05:
I think I figured it out, this solves both problems:
```
lcd.direction(0x60)
sensor.set_vflip(1)
```

Neutree, 2019-04-27:
maybe we need to add new api for `lcd.direction()`, it's not good enough

zpowellman, 2020-03-16:
Hello All,
I have a Maxduino and a Maix Bit. Each is running the latest MaixPY firmware.  I want to set the display to landscape mode.  This code works great on the Maixduino but not on the Maix Bit.

import sensor, lcd
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
#sensor.set_hmirror(1)
sensor.set_vflip(1)
sensor.run(1)
sensor.skip_frames()
lcd.init(freq=15000000)
lcd.direction(lcd.YX_LRUD)  # 0x60)
while(True):
    lcd.display(sensor.snapshot())

I'm using the Maix PY IDE and the board types are set accordingly.  I can't make the Maix bit go into landscape, regardless what I do.  Some of the direction constants do weird things on both boards (i.e. 0x40, 0x80, 0xC0).  Any ideas.  I haven't looked into the code yet, maybe this is the time.  I'm using the latest version of 5.0.  Maybe a mistake?
Thanks,
Dan

Neutree, 2020-08-08:
use `sensor.set_vflip` or `sensor.set_hmirror` or `lcd.rotation` or `lcd.mirror`

closed 2020-08-08 by Neutree