sipeed/MaixPy-v1 #55: core dump: misaligned load

mushroom-x, 2019-04-23:
## Problem 
When I try to use `img.get_statistics()` function,those error message appear occasionally.
## Error Message 
```
core dump: misaligned load
Cause 0x0000000000000004, EPC 0x00000000800fb2e4
reg[00](zero ) = 0x00000000801f8ea9, reg[01](ra   ) = 0x00000000800fb2cc
reg[02](sp   ) = 0x00000000803a8470, reg[03](gp   ) = 0x00000000801c8688
reg[04](tp   ) = 0x0000000000000000, reg[05](t0   ) = 0x0000000000000002
reg[06](t1   ) = 0x0000000000040000, reg[07](t2   ) = 0x0000000000000000
reg[08](s0/fp) = 0x00000000803a8fd0, reg[09](s1   ) = 0x0000000080435870
reg[10](a0   ) = 0x00000000801c74b8, reg[11](a1   ) = 0x000000000000043c
reg[12](a2   ) = 0x0000000080435c9c, reg[13](a3   ) = 0x0000000080435860
reg[14](a4   ) = 0x0000000000000001, reg[15](a5   ) = 0x000000000000043c
reg[16](a6   ) = 0x0000000080436080, reg[17](a7   ) = 0x000000000000006a
reg[18](s2   ) = 0x00000000801f7cb0, reg[19](s3   ) = 0x00000000803a8778
reg[20](s4   ) = 0x0000000000000001, reg[21](s5   ) = 0x00000000801f7b48
reg[22](s6   ) = 0x0000000000000002, reg[23](s7   ) = 0x0000000000002000
reg[24](s8   ) = 0x0000000000000000, reg[25](s9   ) = 0x00000000801f8558
reg[26](s10  ) = 0x0000000000000000, reg[27](s11  ) = 0x0000000000000000
reg[28](t3   ) = 0x000000000000010b, reg[29](t4   ) = 0x0000000000000000
reg[30](t5   ) = 0x0000000000003f40, reg[31](t6   ) = 0x0000000000001720
freg[00](ft0 ) = 0x3f80000100000000(), freg[00]() = 0x00000000801bf9c0()
freg[02](ft2 ) = 0x42c8000000000000(), freg[00]() = 0x00000000801bf9d0()
freg[04](ft4 ) = 0x461c400000000000(), freg[00]() = 0x00000000801bf9e0()
freg[06](ft6 ) = 0x418b6a7900000000(), freg[-1610612736]() = 0x00000000801bf9f0()
freg[08](fs0 ) = 0x467c040000000000(), freg[00]() = 0x00000000801bfa18()
freg[10](fa0 ) = 0x437f000040c00000(), freg[00]() = 0x00000000801bfa48()
freg[12](fa2 ) = 0x3ebd3e1e3ce38e3a(), freg[00]() = 0x00000000801bfa70()
freg[14](fa4 ) = 0x000000004068c7b3(), freg[-536870912]() = 0x00000000801bfa80()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bfa90()
freg[18](fs2 ) = 0x44c5af5000000000(), freg[-536870912]() = 0x00000000801bfab8()
freg[20](fs4 ) = 0x432cb23600000000(), freg[-2147483648]() = 0x00000000801bfac8()
freg[22](fs6 ) = 0x425a082200000000(), freg[00]() = 0x00000000801bfad8()
freg[24](fs8 ) = 0x3b149b9300000000(), freg[-1073741824]() = 0x00000000801bfae8()
freg[26](fs10) = 0x3f80000000000000(), freg[00]() = 0x00000000801bfaf8()
freg[28](ft8 ) = 0x3f80000000000000(), freg[00]() = 0x00000000801bfb20()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x00000000801bfb30()
W (195642323830) SYSCALL: sys_exit called by core 0 with 0x539

```

## Dev Environment
* MaixPy Firmware:  v0.3.0 
* MaixPy Board: Dan Dock
* OS: Ubuntu 18.04

## Code
```
'''
根据背景颜色阈值L通道,筛选出色块.
识别画面中不同颜色的色块及坐标.
把色块从画布坐标系转换为工作台坐标系.

连接maixpy 进入repl
 sudo picocom -b 115200 /dev/ttyUSB0
'''
import sensor
import image
import lcd
import time

# 注1:因为背景是白色，根据白色反射光的原理
# 通过LAB色彩空间的L通道(亮度)对色块进行筛选
# 注2:测试的时候需要外部灯光补光以获得更好的效果．
NOT_BACKGROUND_COLOR = (0, 32, -128, 127, -128, 127)

# BLOCK FEATURE : L_MEAN, A_MEAN, B_MEAN
REF_COLOR_BLOCK_STAT = {
    'GRAY': (41, 10, 6),
    'GREEN': (32, 6, 5),
    'RED': (43, 32, 11),
}
# Init LCD
lcd.init()
# Init Sensor
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
# resolution: 320*240
sensor.set_framesize(sensor.QVGA)
sensor.run(1)

while True:
    img=sensor.snapshot()
    # 设置ROI区域, 不识别Sipeed色块容器
    blobs = img.find_blobs([NOT_BACKGROUND_COLOR], merge=False, area_threshold=400, roi=(0, 0, 320, 180))
    if blobs:
        cnt = 0
        for b in blobs:
            cnt += 1
            print('\n\nBlob: {}\n'.format(cnt))
            # print(b)
            # if b.w()<30 or b.w()>40 or b.h()<30 or b.h()>40 or b.pixels() < 400:
            #     continue
            stat = img.get_statistics(roi=b[0:4])
            print("MEAN VALUE : L:{} A:{} B:{}".format(stat.l_mean(), stat.a_mean(), stat.b_mean()))
            print("MEDIAN VALUE : L:{} A:{} B:{}".format(stat.l_median(), stat.a_median(), stat.b_median()))
            print("MAX VALUE : L:{} A:{} B:{}".format(stat.l_max(), stat.a_max(), stat.b_max()))
            print("MIN VALUE : L:{} A:{} B:{}".format(stat.l_min(), stat.a_min(), stat.b_min()))
            tmp=img.draw_rectangle(b[0:4]) 
            tmp=img.draw_cross(b[5], b[6]) 
            c=img.get_pixel(b[5], b[6])
    lcd.display(img)
```

Neutree, 2019-04-27:
try v0.3.1~~

Neutree, 2020-08-08:
running normally on latest master branch code

closed 2020-08-08 by Neutree

DBGilles, 2021-01-12:
@Neutree what is the cause for this kind of error? I am getting the same error when running a .kmodel using the kendryte standalone SDK. I can't find any information anywhere else

Neutree, 2021-01-12:
@DBGilles  mostly null pointer calling or invalid memory read write

kervel, 2021-04-23:
just for reference if anyone gets the same error ... the solution is here!

https://github.com/kendryte/kendryte-standalone-demo/commit/d6aec51561388551448250718947250beb5e28cc