sipeed/MaixPy-v1 #58: canny edge error:  core dump, breakpoint

mushroom-x, 2019-04-23:
## Problem

use `find_edges()` function, the error message `core dump: breakpoint` appear occasionally.

## Dev Environment

* MaixPy Firmware: v0.2.4
* MaixPy Board: Dan Dock
* OS: Ubuntu 18.04

## Test Code
```python
import sensor
import image
import lcd
import time

# Init LCD
lcd.init()
# Init Sensor
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
# resolution: 320 * 240
sensor.set_framesize(sensor.QVGA)
sensor.run(1)

while True:
    img=sensor.snapshot()
    img_gray = img.to_grayscale(copy=True)
    img_gray.find_edges(image.EDGE_CANNY, threshold=(0,100))
    lcd.display(img_gray)
```
## Error Message

```
core dump: breakpoint
Cause 0x0000000000000003, EPC 0x00000000800d21d8
reg[00](zero ) = 0x0000000000000000, reg[01](ra   ) = 0x00000000800d2048
reg[02](sp   ) = 0x000000008039e580, reg[03](gp   ) = 0x00000000801c42a0
reg[04](tp   ) = 0x0000000000000000, reg[05](t0   ) = 0x000000000000005a
reg[06](t1   ) = 0x00000000000000df, reg[07](t2   ) = 0x0000000000000087
reg[08](s0/fp) = 0x000000008042b650, reg[09](s1   ) = 0x000000008039e880
reg[10](a0   ) = 0x00000000000000de, reg[11](a1   ) = 0x00000000000000dd
reg[12](a2   ) = 0x00000000000116b8, reg[13](a3   ) = 0x0000000000000138
reg[14](a4   ) = 0x000000000000ffff, reg[15](a5   ) = 0x0000000000000000
reg[16](a6   ) = 0x0000000000011580, reg[17](a7   ) = 0x0000000000000138
reg[18](s2   ) = 0x0000000000011580, reg[19](s3   ) = 0x0000000000000064
reg[20](s4   ) = 0x0000000000000140, reg[21](s5   ) = 0x000000000000ffff
reg[22](s6   ) = 0x0000000080471130, reg[23](s7   ) = 0x0000000000000138
reg[24](s8   ) = 0x00000000801c6fa8, reg[25](s9   ) = 0x0000000000000000
reg[26](s10  ) = 0x000000000000001d, reg[27](s11  ) = 0x0000000000012980
reg[28](t3   ) = 0x000000000000007b, reg[29](t4   ) = 0x00000000000000de
reg[30](t5   ) = 0x000000000000002d, reg[31](t6   ) = 0xffffffffffffffff
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb818()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb828()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb838()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb848()
freg[08](fs0 ) = 0x0000000043340000(), freg[00]() = 0x00000000801bb870()
freg[10](fa0 ) = 0x3f80000000000000(), freg[00]() = 0x00000000801bb8a0()
freg[12](fa2 ) = 0x3f80000000000000(), freg[00]() = 0x00000000801bb8c8()
freg[14](fa4 ) = 0x4000000093c42d18(), freg[00]() = 0x00000000801bb8d8()
freg[16](fa6 ) = 0x4000000000000000(), freg[00]() = 0x00000000801bb8e8()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb910()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb920()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb930()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb940()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x00000000801bb950()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x00000000801bb978()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x00000000801bb988()
W (542489468919) SYSCALL: sys_exit called by core 0 with 0x539
```

Neutree, 2019-04-27:
have a try  with v0.3.1?

Neutree, 2020-08-08:
test on latest master branch, running normally

closed 2020-08-08 by Neutree