sipeed/MaixPy-v1 #59: image.binary() doesn't support  rgb565 image

mushroom-x, 2019-04-23:
## Problem
image.binary support grayscale image, but not support rgb565 image format.

## Error Message
```
Traceback (most recent call last):
  File "<stdin>", line 35, in <module>
TypeError: function missing 1 required positional arguments
```
## Code

```python
import sensor
import image
import lcd
import time

BACKGROUND_COLOR = (61, 84, -8, 25, -12, 11)

# Init LCD
lcd.init()
# Init Sensor
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
# resolution: 320 * 240
sensor.set_framesize(sensor.QVGA)
sensor.run(1)

while True:
    img=sensor.snapshot()
    img_gray = img.to_grayscale(copy=True)
    img.binary(thresholds=[BACKGROUND_COLOR])
    lcd.display(img_gray)
```

Neutree, 2019-04-27:
```python
import sensor
import image
import lcd
import time

BACKGROUND_COLOR = (61, 84, -8, 25, -12, 11)

# Init LCD
lcd.init()
# Init Sensor
sensor.reset()
sensor.set_pixformat(sensor.RGB565)
# resolution: 320 * 240
sensor.set_framesize(sensor.QVGA)
sensor.run(1)

while True:
    img=sensor.snapshot()
    img_gray = img.to_grayscale(copy=True)
    img.binary([BACKGROUND_COLOR])
    lcd.display(img_gray)
```

Neutree, 2019-04-27:
missing 1 required positional arguments means need one positional arg, not keyword

closed 2020-08-08 by Neutree

Ali1flayyih, 2021-02-23:
#try this code:

import sensor, image, time

sensor.reset()
sensor.set_pixformat(sensor.RGB565) 
sensor.set_framesize(sensor.QVGA)
sensor.skip_frames(time = 2000)
clock = time.clock()

BACKGROUND_COLOR = (61, 84, -8, 25, -12, 11)

while True:
    clock.tick()
    img = sensor.snapshot()
    mask_red_oragne = img.binary([BACKGROUND_COLOR]).erode(1).to_bitmap(copy=True)
    print("FPS %f" % clock.fps())