sipeed/MaixPy-v1 #60: add sensor.alloc_extra_fb

mushroom-x, 2019-04-23:
add `alloc_extra_fb` function  support for in memeory basic frame difference.

Neutree, 2019-04-27:
maybe you can use `image.Image()` now, this create a image with size 320x240 and RGB565 format

edward687, 2019-05-26:
> maybe you can use `image.Image()` now, this create a image with size 320x240 and RGB565 format
God, How to rewrite the following code with 
`extra_fb = sensor.alloc_extra_fb(sensor.width(), sensor.height(), sensor.RGB565)
extra_fb.replace(sensor.snapshot())`

Milan991, 2020-10-12:
@Neutree Can you send me an example of `image.Image` usage because when I'm using as `extra_fb = image.Image(size=(320, 240))` I'm getting out of heap memory error.

Even when I lower the resolution to 176, 120 error remains.

My goal is to use this openMV example: in_memory_basic_frame_differencing.py

Board is MaixCube.

Neutree, 2020-10-14:
@Milan991  you can change gc size bigger by 
```
from Maix import utils
utils.gc_heap_size(new_size)
```
default is 512*1024

Milan991, 2020-10-14:
@Neutree Thank you. I will try it. What is the maximum size I can set with MaixCube?

Milan991, 2020-10-14:
@Neutree I've given it a test but without success. Still getting the error even when heap size is increased to 2048*1024. The function that creates this error is `img.get_histogram()`. When I remove that function everything works well.

Code:
```
# In Memory Basic Frame Differencing Example
#
# This example demonstrates using frame differencing with your OpenMV Cam. It's
# called basic frame differencing because there's no background image update.
# So, as time passes the background image may change resulting in issues.

import sensor, image, os
from Maix import utils

TRIGGER_THRESHOLD = 5

utils.gc_heap_size(2048 * 1024)

sensor.reset()                      # Reset and initialize the sensor. It will
                                    # run automatically, call sensor.run(0) to stop
sensor.set_pixformat(sensor.RGB565) # Set pixel format to RGB565 (or GRAYSCALE)
sensor.set_framesize(sensor.QQQQVGA)   # Set frame size to QVGA (320x240)
sensor.skip_frames(time = 2000)     # Wait for settings take effect.
clock = time.clock()                # Create a clock object to track the FPS.

print("About to save background image...")
gc.enable()
extra_fb = image.Image(size=(40, 30), copy_to_fb = False)
sensor.skip_frames(time = 2000) # Give the user time to get ready.
extra_fb.replace(sensor.snapshot())
print("Saved background image - Now frame differencing!")

while(True):
    clock.tick() # Track elapsed milliseconds between snapshots().
    img = sensor.snapshot() # Take a picture and return the image.
    # Replace the image with the "abs(NEW-OLD)" frame difference.
    img.difference(extra_fb)

    hist = img.get_histogram()
    # This code below works by comparing the 99th percentile value (e.g. the
    # non-outlier max value against the 90th percentile value (e.g. a non-max
    # value. The difference between the two values will grow as the difference
    # image seems more pixels change.
    diff = hist.get_percentile(0.99).l_value() - hist.get_percentile(0.90).l_value()
    triggered = diff > TRIGGER_THRESHOLD

    print(clock.fps(), triggered, gc.mem_free()) # Note: Your OpenMV Cam runs about half as fast while
    # connected to your computer. The FPS should increase once disconnected.
    gc.collect()
```