sipeed/MaixPy-v1 #63: UART constructor parity can't be None

mushroom-x, 2019-04-24:
## Problem
UART constructor parity can't be None  . It's conflict with documents [machine.UART](https://maixpy.sipeed.com/zh/libs/machine/uart.html)

```
parity: 奇偶校验位，支持 None, machine.UART.PARITY_ODD, machine.UART.PARITY_EVEN （默认的 REPL 使用的串口（UARTHS）只支持 None）， 默认 None
```

##  Code Segments
```python
uart2 = UART(UART.UART2, 115200, 8, None, 1, timeout=1000, read_buf_len=4096)
```
**Error Message**
```
Traceback (most recent call last):
  File "<stdin>", line 17, in <module>
TypeError: can't convert NoneType to int
```

***Error Message**
```python
>>> uart = UART(UART.UART2)
>>> uart.
__class__       __next__        UART1           UART2
UART3           UARTHS          deinit          init
read            readinto        readline        write
>>> uart.init(115200, 8, None, 1, timeout=1000, read_buf_len=4096)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: can't convert NoneType to int

```

## Dev Environment

 * MaixPy Firmware: v0.2.4
 * MaixPy Board: Dan Dock
 * OS: Ubuntu 18.04

mushroom-x, 2019-04-24:
only this way is work:
```python
uart2 = UART(UART.UART2)
uart2.init()
```
because the parameter is same as the default config in firmware: 
[machine_uart_make_new](https://github.com/sipeed/MaixPy/blob/b3bd57bfdcd3069e73c9ed3da520ab81841d2050/ports/k210-freertos/mpy_support/standard_lib/machine/machine_uart.c)

```c
STATIC mp_obj_t machine_uart_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) {
    mp_arg_check_num(n_args, n_kw, 1, MP_OBJ_FUN_ARGS_MAX, true);
    // get uart id
    mp_int_t uart_num = mp_obj_get_int(args[0]);
    if (uart_num < 0 || UART_DEVICE_MAX == uart_num) {
        nlr_raise(mp_obj_new_exception_msg_varg(&mp_type_ValueError, "[MAIXPY]UART%b:does not exist", uart_num));
    }else if (uart_num > MICROPY_UARTHS_DEVICE) {
    	nlr_raise(mp_obj_new_exception_msg_varg(&mp_type_ValueError, "[MAIXPY]UART%b:does not exist", uart_num));
    }
    // create instance
    machine_uart_obj_t *self = m_new_obj(machine_uart_obj_t);
    self->base.type = &machine_uart_type;
    self->uart_num = uart_num;
    self->baudrate = 0;
    self->bitwidth = 8;
    self->parity = 0;
    self->stop = 1;
	self->read_buf_len = 0;
	self->data_len = 0;
    mp_map_t kw_args;
    mp_map_init_fixed_table(&kw_args, n_kw, args + n_args);
    machine_uart_init_helper(self, n_args - 1, args + 1, &kw_args);
    return MP_OBJ_FROM_PTR(self);
}
```

mushroom-x, 2019-04-24:
pass in extra parameters in constructor or init function ,  `TypeError` arise.
```
>>> uart2 = UART(UART.UART2)
>>> uart2.init(boundrate=115200, bits=8, stop=1, timeout=1000, read_buf_len=4096)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: extra keyword arguments given

```

Neutree, 2019-04-26:
upgrade to v0.3.0 please

Neutree, 2019-04-26:
baudrate but not boundrate

closed 2020-08-08 by Neutree