sipeed/MaixPy-v1 #65: MaixPy IDE for macOS?

dongzhixia, 2019-04-29:
Hi, is there a macOS build available?

thepian, 2019-04-29:
You can build using Docker, I did a dockerfile based on `buildpack-deps:stretch`, it's fairly straight forward.

You might also get it to work straight in macOS mine bombed out in a build script that is probably hardcoded to Linux.

Neutree, 2019-04-30:
we have no macOS currently..... so... maybe later

win32cpp, 2019-04-30:
@Neutree I would like to talk to you. Please inform me about your communication.(telegram-whatsapp-facebook-...) .

Neutree, 2019-05-08:
download here: http://dl.sipeed.com/MAIX/MaixPy/ide/v0.2.2/

dongzhixia, 2019-05-08:
@Neutree  :thumbsup:

thepian, 2019-05-14:
I've downloaded this and tried it on all three boards. I see nothing in the framebuffer panes, and cannot connect to the device.

KmiloRod, 2019-08-22:
> I've downloaded this and tried it on all three boards. I see nothing in the framebuffer panes, and cannot connect to the device.

Same problem here. I'm trying to connect to a Maix Bit with Mic with the last firmware installed (0.3.2) using MaixPy IDE 0.2.4 for macOS, and the Connection dialog just keeps showing "Connecting... Please wait..." without success.

KmiloRod, 2019-08-23:
> @KmiloRod http://dl.sipeed.com/MAIX/MaixPy/ide/v0.2.4/readme.txt

Thank you. I loaded firmware 0.4.0_44 and the IDE is working now.

mattkanwisher, 2020-07-08:
Seems to crash instantly on OS X Catalina. Anyone else have a problem?

whatcall, 2020-07-22:
> Seems to crash instantly on OS X Catalina. Anyone else have a problem?

Yes, macOS 10.14 crashes too.
I asked help from QQ chat group and they told me the latest release supports up to macOS 10.13.

Neutree, 2020-08-08:
@whatcall download here: https://dl.sipeed.com/MAIX/MaixPy/ide/v0.2.5 , it compiled on catalina, if problems exist, reopen this issue or create new one

closed 2020-08-08 by Neutree