sipeed/MaixPy-v1 #68: usocket lack of socket.bind, socket.listen and socket.accept

rychong, 2019-04-30:
The usocket in Maixpy lack of functions such as socket.bind, socket.listen and socket.accept, thus Maixpy can't work as a server.
More over, other package like upip etc. is still absense, we hope maixpy can support and integrate more drivers and package.

mchuv, 2020-06-09:
After 1 year (with the 0.5 firmware version), Maixpy still doesn't have support for socket servers, I think like rychong, that it would be a very good new to have this options.
I'm try to create a product using the Maix Dock, and I need to have a small WEB server running in the Maix Dock, this server would show a configuration page, to setup de WIFI SID, password, AND the IP of a server to send data acquired by the MaixPy.

alessandro-montanari, 2020-12-26:
I also need this functionality. 
Looking forward to the fix!

JohnsonLuo-2019, 2021-03-30:
MaixPy uses SPI to communicate with ESP32. K210 is SPI master, ESP32 is SPI slave. There is no way for slave ESP32 to interrupt master K210. This is why MaixPy does not implement bind/listen/... server functionality.