sipeed/MaixPy-v1 #77: 设置CPU主频为26mhz之后，键盘输入无效

backahasten, 2019-05-12:
设置CPU主频为26mhz之后，键盘输入无效，刷也刷不掉，需要手动改写0x600000位置的信息，很不方便，建议增加全盘擦除功能

Neutree, 2019-05-13:
代码里不会增加全盘擦除的功能，如果需要擦除全盘请生成全是`0xFF`的.bin，打包`kfpkg`烧录即可

另外这个地址看来是非常不安全的，等下我会修改这个地址

Neutree, 2019-05-13:
26MHz恐怕不能设置115200的波特率了，得用9600了

Neutree, 2019-05-13:
我再最新的代码中更新了这个问题：
* 如果在串口`REPL`不能使用的情况下需要清空文件系统，使用[kflash_gui](https://github.com/sipeed/kflash_gui) 烧录 [erase_spiffs.kfpkg](http://dl.sipeed.com/MAIX/MaixPy/release/erase_spiffs.kfpkg)可以了
* 删除了`cpufreq`这个模块，移动到了`Maix.freq`，详细文档见[Maix.freq](https://maixpy.sipeed.com/zh/libs/Maix/freq.html)
* 当设置的频率低于`600MHz`时串口`REPL`波特率自动设置为`9600`而不是`115200`

backahasten, 2019-05-13:
600Mhz??emmmm，写错了把

Neutree, 2019-05-13:
没写错， 在`MaixPy`中`CPU`限制最高支持`600MHz`

Neutree, 2019-05-13:
是的， 是低于 `60MHz`时

Neutree, 2020-08-08:
现在也可以用 kflash_gui 对flash 或文件系统进行擦除

closed 2020-08-08 by Neutree