sipeed/MaixPy-v1 #79: Any AI examples in the MaixPy library?

Tylersuard, 2019-05-13:
Hello!  I love the price and capabilities of the MAIX, seriously considering getting one.  My main language is Micropython, I'm not seeing any machine learning demos in your examples folder for the micropython version?  I was hoping to see some examples that could detect faces, detect if someone is smiling, etc.  Thanks!

Neutree, 2019-05-13:
now we haven't provide so much model, you can try to train the model yourself. and then load it with maixpy to run

Neutree, 2019-05-13:
like [hardware.ai](https://www.youtube.com/watch?v=fZ4QbtAjKXo) ( [article](https://www.instructables.com/id/Transfer-Learning-With-Sipeed-MaiX-and-Arduino-IDE/) ) did

Neutree, 2020-08-08:
face recognition: https://github.com/sipeed/MaixPy_scripts/blob/master/machine_vision/demo_face_recognition.py
find face: https://github.com/sipeed/MaixPy_scripts/blob/master/machine_vision/demo_find_face.py
more: https://github.com/sipeed/MaixPy_scripts/tree/master/machine_vision
more models: https://www.maixhub.com/

closed 2020-08-08 by Neutree