sipeed/MaixPy-v1 #80: Max sensor resolution

albydnc, 2019-05-13:
I've tried to set VGA resolution on the Maix Bit but it hangs and disconnect from the pc. Image is too big for the buffer.
In the docs it says it is supported.

Neutree, 2019-05-14:
not support VGA yet for we want to leave more memory

Neutree, 2019-05-14:
I add config for resolution change here:

https://github.com/sipeed/MaixPy/blob/2d64fc6614fb74729595866ac66a7442123742f1/ports/k210-freertos/platform/drivers/include/driver_config.h#L6-L10

change to 
```c
#define LCD_W_MAX 640
#define LCD_H_MAX 480

#define LCD_W     640
#define LCD_H     480
```

alessandro-montanari, 2019-05-26:
I'd like to take pictures at VGA resolution as well. On the forum it was suggested that maybe we could get a new binary with the buffer size increased: https://bbs.sipeed.com/t/topic/787
That would be great! :)

amckay1, 2020-04-26:
I would also like to be able to increase the buffer further to at least allow the `dual_buff = True` option for VGA video (currently getting out of memory error) at expense of ML models. My goal is to get an optimal balance of fps, resolution, jpeg quality, and save to the SD card. I tried unsuccessfully to follow the instructions above (compile from source after increasing resolution in `driver_config.h`) but given the instructions are from May I'm hoping there is a better solution now. Thank you!

Neutree, 2020-08-08:
@amckay1 now max lcd width and height config by `python3 project.py menuconfig`, add `dual_buff = True` will increase fps, but need more about 384KiB memory

if remain some problems, reopen it or create a new issue

closed 2020-08-08 by Neutree