sipeed/MaixPy-v1 #81: Unable to compile MaixPy

keithr0, 2019-05-15:
I am attempting to compile MaixPy using an Ubuntu image in Windows Subsystem for Linux under windows 10. I get the following errors

`/mnt/c/ubuntu/MaixPy/ports/k210-freertos/mpy_support/standard_lib/include/vfs_spiffs.h:95:52: error: 'fs' undeclared here (not in a function); did you mean 'fls'?
 extern u8_t spiffs_work_buf[SPIFFS_CFG_LOG_PAGE_SZ(fs)*2];
                                                                                         ^~
                                                                                         fls`

`In file included from standard_lib/uos/vfs_spiffs_io.c:7:
/mnt/c/ubuntu/MaixPy/ports/k210-freertos/mpy_support/standard_lib/include/vfs_spiffs.h:95:29: warning: implicit declaration of function 'SPIFFS_CFG_LOG_PAGE_SZ'; did you mean 'SPIFFS_ERR_NOT_A_FS'? [-Wimplicit-function-declaration]
 extern u8_t spiffs_work_buf[SPIFFS_CFG_LOG_PAGE_SZ(fs)*2];
                                           ^~~~~~~~~~~~~~~~~~~~~~
                                            SPIFFS_ERR_NOT_A_FS`

Neutree, 2019-05-15:
not test in windows subsystem , please use ubuntu instead

keithr0, 2019-05-16:
I attempted to build using Ubuntu in a docker container, but the result is the same.

Neutree, 2019-05-16:
https://github.com/sipeed/MaixPy/blob/34774e17c860819bc080a3bb2c091f0cfb3aad3c/.travis.yml#L25-L34

https://travis-ci.org/sipeed/MaixPy

keithr0, 2019-05-19:
I eventually got a good compile with this script:

```
#!/bin/bash
apt-get update
apt-get install -y sshpass
git clone --recursive https://github.com/sipeed/MaixPy.git
wget https://github.com/kendryte/kendryte-gnu-toolchain/releases/download/v8.2.0-20190409/kendryte-toolchain-ubuntu-amd64-8.2.0-20190409.tar.xz
xz -dk kendryte-toolchain-ubuntu-amd64-8.2.0-20190409.tar.xz && tar -xf kendryte-toolchain-ubuntu-amd64-8.2.0-20190409.tar -C /opt
cd MaixPy
make -C mpy-cross
cd ports/k210-freertos
echo -e "toolchain_path=/opt/kendryte-toolchain\nbaud=2000000\ndevice=/dev/ttyUSB0\nBoard=dan\n" > config.conf
chmod +x build.sh && ./build.sh
cd output

```
There were a lot of warnings though.

closed 2020-08-08 by Neutree