sipeed/MaixPy-v1 #82: Problem running Face detector example

keithr0, 2019-05-15:
I am having problems running this example on Maix Go, the error is in this line

`task = kpu.load(0x300000)`

```
v=596458738, flag=-2096411809, arch=-1955282835, layer len=834782759, mem=-392490997, out cnt=-663129117
err: we only support V3 now, get V596458738
[MAIXPY]kpu: kpu_model_get_size error -3
```

```
ValueError: [MAIXPY]:kpu load error: read model size error 1
MicroPython v0.3.1 on 2019-04-27; Sipeed_M1 with kendryte-k210
```

Neutree, 2019-05-15:
you just load the wrong model, ensure you have burned right model to 0x300000

keithr0, 2019-05-15:
I downloaded Face.kpfg from 
`http://dl.sipeed.com/MAIX/MaixPy/model/face.kfpkg`

Is there a different location or an upgraded file?

Neutree, 2019-05-15:
download this one : https://github.com/sipeed/MaixPy/releases/download/v0.3.2/face_model_at_0x300000.kfpkg

keithr0, 2019-05-15:
That fixed it thank you.

closed 2019-05-15 by keithr0

kimchicanh, 2019-05-17:
@keithr0 
hi,
how to burn model to 0x300000
I am just a newbie

thanks