sipeed/MaixPy-v1 #83: compile error with CMAKE_C_COMPILER

kimchicanh, 2019-05-16:
Hi,
I run ./build.sh and get this error:
make -C sdk/ compile || exit "$?"; make -C drivers/ compile || exit "$?"; make -C utils/ compile || exit "$?"; make -C api/ compile || exit "$?";
make[2]: Entering directory '/home/Code/MaixPy/ports/k210-freertos/platform'
make[2]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
if [ ! -d /home/Code/MaixPy/ports/k210-freertos/platform/sdk/kendryte-standalone-sdk ] ; \
then \
     git clone -b Maixpy-dev https://github.com/sipeed/kendryte-standalone-sdk.git;\
fi;
mkdir -p build
cd build && cmake ../kendryte-standalone-sdk/ -DPROJ=hello_world -DTOOLCHAIN=/opt/kendryte-toolchain//bin && make -j32
PROJ = hello_world
-- Check for RISCV toolchain ...
-- Using /opt/kendryte-toolchain//bin RISCV toolchain
CMake Error at cmake/toolchain.cmake:50 (message):
  CMAKE_C_COMPILER is not in kendryte-toolchain dist/bin folder.
Call Stack (most recent call first):
  cmake/common.cmake:30 (include)
  CMakeLists.txt:16 (include)

This is my config:
toolchain_path=/opt/kendryte-toolchain/
baud=2000000
device=/dev/ttyUSB0
Board=dan

I try to export:
export CMAKE_C_COMPILER=/opt/kendryte-toolchain/bin/riscv64-unknown-elf-gcc
export CMAKE_CXX_COMPILER=/opt/kendryte-toolchain/bin/riscv64-unknown-elf-g++
export CMAKE_LINKER=/opt/kendryte-toolchain/bin/riscv64-unknown-elf-ld
export CMAKE_OBJCOPY=/opt/kendryte-toolchain/bin/riscv64-unknown-elf-objcopy
export CMAKE_OBJDUMP=/opt/kendryte-toolchain/bin/riscv64-unknown-elf-objdump

But it do not work
Please help me

Thanks

Neutree, 2019-05-16:
from your log, I didn't understand where the error is!

compile looks like this:
```shell
sudo apt-get update
sudo apt-get install build-essential cmake
wget https://github.com/kendryte/kendryte-gnu-toolchain/releases/download/v8.2.0-20190409/kendryte-toolchain-ubuntu-amd64-8.2.0-20190409.tar.xz
xz -dk kendryte-toolchain-ubuntu-amd64-8.2.0-20190409.tar.xz && tar -xf kendryte-toolchain-ubuntu-amd64-8.2.0-20190409.tar -C /opt
cd MaixPy
make -C mpy-cross
cd ports/k210-freertos
echo -e "toolchain_path=/opt/kendryte-toolchain\nbaud=2000000\ndevice=/dev/ttyUSB0\nBoard=dan\n" > config.conf
chmod +x build.sh && ./build.sh

```

and the compile log you can refer [here](https://travis-ci.org/sipeed/MaixPy)

israelpz, 2019-05-28:
> Hi,
> I run ./build.sh and get this error:
> make -C sdk/ compile || exit "$?"; make -C drivers/ compile || exit "$?"; make -C utils/ compile || exit "$?"; make -C api/ compile || exit "$?";
> make[2]: [...]

Remove last slash from:
toolchain_path=/opt/kendryte-toolchain

closed 2020-08-08 by Neutree