sipeed/MaixPy-v1 #92: rtc not working

ezu00261, 2019-05-29:
rtc not working on Maix Go.

```
import time

time.set_time((2019,1,1,0,0,0))

for i in range(1,10):
    print(time.localtime())
    time.sleep(1)
```

result
```
True
(0, 0, 29, 14, 56, 46, 5, 29)
(0, 0, 10, 10, 23, 36, 0, 10)
(0, 0, 23, 8, 36, 42, 6, 23)
(0, 0, 4, 6, 49, 16, 1, 4)
(0, 0, 17, 5, 3, 26, 0, 17)
(0, 0, 30, 3, 16, 6, 6, 30)
(0, 0, 11, 1, 29, 56, 1, 11)
(0, 0, 23, 23, 43, 26, 6, 23)
(0, 0, 4, 21, 56, 36, 1, 4)
MicroPython v0.3.2 on 2019-05-15; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>>
```

Neutree, 2019-05-31:
bug from kendryte SDK: https://github.com/kendryte/kendryte-standalone-sdk/commit/019b841f7d3c893e2b8af95840c723bd6ac4876f

fix in next version

ezu00261, 2019-06-04:
I try MaixPy(ea8cd88) with sipeed/kendryte-standalone-sdk@50676ff.
still issue not resolved.

```
(2000, 0, 0, 0, 8, 26, 3, 0)
(2000, 0, 13, 9, 17, 6, 2, 13)
(2000, 0, 26, 21, 56, 33, 1, 26)
(2000, 0, 8, 10, 36, 3, 4, 8)
(2000, 0, 21, 16, 2, 54, 3, 21)
(2000, 0, 3, 4, 42, 26, 6, 3)
(2000, 0, 16, 10, 9, 14, 5, 16)
(2000, 0, 29, 22, 48, 56, 4, 29)
(2000, 0, 11, 11, 29, 56, 0, 11)
MicroPython v0.3.2-15-gea8cd88cf on 2019-06-02; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
```

but under clocking cpu (416->403MHz), seems ok.
this issue seems to be related to the frequency of cpu(pll0).

```
(2019, 1, 1, 0, 0, 0, 2, 1)
(2019, 1, 1, 0, 0, 1, 2, 1)
(2019, 1, 1, 0, 0, 2, 2, 1)
(2019, 1, 1, 0, 0, 3, 2, 1)
(2019, 1, 1, 0, 0, 4, 2, 1)
(2019, 1, 1, 0, 0, 5, 2, 1)
(2019, 1, 1, 0, 0, 6, 2, 1)
(2019, 1, 1, 0, 0, 7, 2, 1)
(2019, 1, 1, 0, 0, 8, 2, 1)
MicroPython v0.3.2-15-gea8cd88cf on 2019-06-02; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>> import Maix
>>> Maix.freq.get_cpu()
403

```

Neutree, 2020-09-16:
yes it's a hardware bug! now default frequency changed to 400MHz, it works now

closed 2020-09-16 by Neutree