sipeed/MaixPy-v1 #97: crop image

kimchicanh, 2019-06-03:
Hi,
I want to crop image for face recognition
img = sensor.snapshot()
code = kpu.run_yolo2(task, img)
    if code:
        for i in code:
            a = img.draw_rectangle(i.rect())
I try img.crop but my program is broke
How I can crop image and get it to another task, I am just a newbie
Thanks

syakujiimati, 2019-09-05:
Hello!
I had the same problem.
When I passed the face coordinates to copy as shown below, it worked.
    while(True):
        img = sensor.snapshot() # Take an image from sensor
        bbox = kpu.run_yolo2(task, img) # Run the detection routine
        if bbox:
            for i in bbox:
                print(i)
                #img.draw_rectangle(i.rect())
                img = img.copy(i.rect())
        lcd.display(img)

I used Google translation. Sorry if it was a funny sentence

Neutree, 2020-08-08:
`image.copy` do

Neutree, 2020-08-08:
and pay attention, if you use the croped image on kpu, you shoul call `img.pix_to_ai()` before call `kpu.forward`

closed 2020-08-08 by Neutree