sipeed/MaixPy-v1 #103: IDE BUG: When a boot.py has a problem, IDE can't connect

laurentopia, 2019-06-08:
repro:
1. install the content of this repo to the maix bit https://github.com/laurentopia/AI-Avoidance
= when the boot.py reboots it'll hang
3. launche MAIXPY IDE and try to connect
= you can't it keeps looping

seedsofhope, 2019-08-04:
What do you mean boot has a problem?
Ide is not connecting for me anymore... Putty connects, ide does not, on both windows and Ubuntu. My board is maixduino with esp32. Firmware 0.3.2 minimal...
... Sipeed guys, plz fix the ide

daryllman, 2020-03-23:
@seedsofhope have u solved ur problem? I faced the exact same problem...! help pls

laurentopia, 2020-03-23:
no i gave up on their stuff

daryllman, 2020-03-24:
@laurentopia @seedsofhope Just in case you guys are interested and others who visit this thread, what solved the problem for me was to use maixpy firmware **v0.4.0** and above. This solution should be more pertinent to @seedsofhope. The MaixPy IDE would only connect if its this version and above. Found this statement somewhere after diving into the documentation and other resources for quite some time, was not stated explicitly.

You can find the repo of all the firmware versions here:
[https://dl.sipeed.com/MAIX/MaixPy/release/master/](https://dl.sipeed.com/MAIX/MaixPy/release/master/)

closed 2020-08-08 by Neutree