sipeed/MaixPy-v1 #105: SSL for Maix Go

jayeheffernan, 2019-06-12:
I'm trying to send HTTPS requests from a script on a Maix Go.  It seems there is no support for SSL: the `usocket` library's `IPPROTO_*` constants are missing, and there is no module named `ssl` or `ussl`.

Does the Maix Go support SSL?  If not, will support be added to firmware soon?

Is there an SSL library I can copy to the device, to fill in for the missing `ussl` library?

TakuroFukamizu, 2019-07-21:
I need it too...
I hope the ussl module included to the  maixpy in the future...

AIWintermuteAI, 2019-10-26:
Any progress on that one? As people pointed, without HTTPS support MaixPy board cannot be used as a “true” IoT device, because right now IoT clouds like Azure, AWS, Google and others require https connection using ssl.

lypanov, 2020-01-02:
Any news on this? Massive disappointed to purchase one of these boards in time for my programming-vacation only to realize I can't actually even begin the project without first migrating the board to https://github.com/loboris/MicroPython_K210_LoBo

updesh31, 2020-01-08:
For this I used Adruino ide to program esp8285 and now i am able to send image from Maixpy board to Telegram using SSL.

valenzmanu, 2020-05-27:
@updesh31  How did you do it? https://github.com/sipeed/MaixPy/issues/105#issuecomment-572131828

updesh31, 2020-05-27:
share me your mail id i will share both codes for maixpy and wifi......:)

junhuanchen, 2020-05-28:
The implementation is here, not merged yet.
https://github.com/junhuanchen/MaixPy/commit/ec7506219f7ebd14bac711c8d48d6e4c5bc87191
You can try it first.

valenzmanu, 2020-05-28:
Thanks @updesh31! mantt89.mv@gmail.com

junhuanchen, 2020-07-01:
Latest supported.

closed 2020-07-01 by junhuanchen

Vigneshprabhu97, 2020-09-18:
@junhuanchen Please help me with installing the maixpy ide which supports SSL ,ussl modules

valenzmanu, 2020-09-22:
@Vigneshprabhu97 I was able to use ussl modules with the firmware in this branch: <br>https://github.com/sipeed/MaixPy/tree/develop

You can follow the build instructions [here](https://github.com/sipeed/MaixPy/blob/develop/build.md). Hope you find it useful.

junhuanchen, 2020-09-22:
build firmware / master enable ussl modules

Vigneshprabhu97, 2020-09-24:
> @Vigneshprabhu97 I was able to use ussl modules with the firmware in this branch:
> https://github.com/sipeed/MaixPy/tree/develop
> 
> You can follow the build instructions [here](https://github.com/sipeed/MaixPy/blob/develop/build.md). Hope you find it useful.
> 
> I used the same MaixPy IDE btw.

If You have any maixpy scripts for IoT please do share me vigneshprabhu156@gmail.com

ReginaldSourn, 2021-08-16:
> share me your mail id i will share both codes for maixpy and wifi......:)

Please @updesh31 I need your solutions too.