sipeed/MaixPy-v1 #110: KPU run_yolo2 from SD card image

wizath, 2019-06-27:
I want to try KPU with image sourced from SD card. But when I load image
```
>>> x = image.Image('/sd/1.jpg')
>>> code = kpu.run_yolo2(task, x)
[MAIXPY]kpu: pix_ai is NULL!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: [MAIXPY]kpu: check img format err!
>>> x
{"w":320, "h":240, "type"="rgb565", "size":153600}
```

Anyway, when I save image from sensor.snapshot(), then load it, the error is exactly the same
```
>>> img = sensor.snapshot()
>>> img.save('/sd/1.jpg')
{"w":320, "h":240, "type"="rgb565", "size":153600}
>>> x = image.Image('/sd/1.jpg')
>>> code = kpu.run_yolo2(task, x)
[MAIXPY]kpu: pix_ai is NULL!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: [MAIXPY]kpu: check img format err!
```

Does anyone have any hint what could be wrong?

Vinschni, 2019-06-30:
I tried to generate images / mjpeg data with ffmpeg as input for kpu yolo2 but were not successful.
`mediainfo` of a avi recorded to SD gives me:
```
Complete name                            : capture.avi
Format                                   : AVI
Format/Info                              : Audio Video Interleave
File size                                : 470 KiB

Video
ID                                       : 0
Format                                   : JPEG
Codec ID                                 : MJPG
Width                                    : 320 pixels
Height                                   : 240 pixels
Display aspect ratio                     : 4:3
Frame rate                               : 4.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:2
Bit depth                                : 8 bits
Compression mode                         : Lossy
```

wizath, 2019-06-30:
As I was searching the source code for any hint, it seems to be quite reasonable.

We have image_t struct with this weird pix_ai field.
https://github.com/sipeed/MaixPy/blob/a21a2ba1c305252543d43a512ace708a9413ac00/ports/k210-freertos/mpy_support/omv/img/include/imlib.h#L418

Then in Maix_kpu there is this check
https://github.com/sipeed/MaixPy/blob/a21a2ba1c305252543d43a512ace708a9413ac00/ports/k210-freertos/mpy_support/Maix/Maix_kpu.c#L70

TL;DR Loading images from file will read image data only to _data field_, not the _pix_ai_ and that is the cause of that validation error.

But now how to copy image contents (and which format) to the pix_ai field of image struct? According to comment this is for _speedup_.

Vinschni, 2019-07-02:
I found a function which probably does what you need:
https://github.com/sipeed/MaixPy/blob/f075d9f8a336829826dd4e6eb05d9a43a0dbf689/ports/k210-freertos/mpy_support/omv/py/py_image.c#L5577

call it like `img.pix_to_ai()` to use it.

wizath, 2019-07-02:
Thanks! Overlooked it, it now works fine, except some heap issues :champagne:

closed 2019-07-02 by wizath

DeqingSun, 2020-04-29:
> I found a function which probably does what you need:
> https://github.com/sipeed/MaixPy/blob/f075d9f8a336829826dd4e6eb05d9a43a0dbf689/ports/k210-freertos/mpy_support/omv/py/py_image.c#L5577
> 
> call it like `img.pix_to_ai()` to use it.

Thanks, although this may cause heap memory leak. 
using ```del img``` can release the heap memory.

lastrei, 2020-06-09:
> > I found a function which probably does what you need:
> > https://github.com/sipeed/MaixPy/blob/f075d9f8a336829826dd4e6eb05d9a43a0dbf689/ports/k210-freertos/mpy_support/omv/py/py_image.c#L5577
> > 
> > call it like `img.pix_to_ai()` to use it.
> 
> Thanks, this helped me as well. Although this may cause heap memory leak, using `del img` can release the heap memory.

but when i use pix_to_ai still return none?

may i see your code ?

wiebevr, 2020-11-25:
`pix_to_ai()` works in place. The following works for me:
```
img = image.Image('/flash/image.jpg').to_rgb565(copy=True)
img.pix_to_ai()
```