sipeed/MaixPy-v1 #112: Build Error:  v0.3.2 , undefined reference

xiangweizeng, 2019-06-30:
/root/MaixPy/ports/k210-freertos/mpy_support/Maix/Maix_kpu.c:1239: undefined reference to `kpu_run_kmodel'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /root/MaixPy/ports/k210-freertos/mpy_support/Maix/Maix_kpu.c:1244: undefined reference to `kpu_get_output'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /root/MaixPy/ports/k210-freertos/output/mpy_support.a(Maix_kpu.o): in function `py_kpu_class_run_yolo2':
/root/MaixPy/ports/k210-freertos/mpy_support/Maix/Maix_kpu.c:833: undefined reference to `kpu_run_kmodel'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /root/MaixPy/ports/k210-freertos/output/mpy_support.a(Maix_kpu.o): in function `py_kpu_class_load':
/root/MaixPy/ports/k210-freertos/mpy_support/Maix/Maix_kpu.c:334: undefined reference to `kpu_load_kmodel'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /root/MaixPy/ports/k210-freertos/output/api.a(sipeed_yolo2.o): in function `region_layer_init':
/root/MaixPy/ports/k210-freertos/platform/api/sipeed_yolo2.c:54: undefined reference to `kpu_get_output'
/opt/kendryte-toolchain/lib/gcc/riscv64-unknown-elf/8.2.0/../../../../riscv64-unknown-elf/bin/ld: /root/MaixPy/ports/k210-freertos/output/api.a(sipeed_conv.o): in function `sipeed_conv_run':
/root/MaixPy/ports/k210-freertos/platform/api/sipeed_conv.c:221: undefined reference to `kpu_run'
collect2: error: ld returned 1 exit status

robert-hh, 2019-06-30:
Maybe a wrong version of lib_sipeed_kpu.a. On my system building works fine. The ..../ports/k210-freertos/platform/api/lib_sipeed_kpu.a  has the following entry:

`71512 Jun 19 07:35 lib_sipeed_kpu.a`

Neutree, 2019-06-30:
use the master branch will be ok

closed 2020-08-08 by Neutree