sipeed/MaixPy-v1 #113: SPI driver unstable

Vinschni, 2019-07-01:
Hello together,
writing data to SD is not robust with MaixPy neither with commit 4cca70fdb8191bda9959f631db1a12170b556a68 nor with commit beae8648307f27a693be3ebc9f2e169f7f09627d
(i tried both).

After a certain amount of data the SPI driver fails reproducably.
Use case is continuous logging to SD card, every 1000ms up to 200 bytes are logged to a file.

Result is a crash:

```
core dump: misaligned load
Cause 0x0000000000000004, EPC 0x00000000800afc42
reg[00](zero ) = 0x0000000053000000, reg[01](ra   ) = 0x00000000800afbea
reg[02](sp   ) = 0x00000000803599c0, reg[03](gp   ) = 0x0000000080187200
reg[04](tp   ) = 0x0000000000000000, reg[05](t0   ) = 0x0000000000000000
reg[06](t1   ) = 0x0000000000010000, reg[07](t2   ) = 0x0000000000000013
reg[08](s0/fp) = 0x0000000080543680, reg[09](s1   ) = 0x0000000000000000
reg[10](a0   ) = 0x0000000080543680, reg[11](a1   ) = 0x00000000801877e8
reg[12](a2   ) = 0x0000000000000400, reg[13](a3   ) = 0x00000000803d7690
reg[14](a4   ) = 0x00000000801dc89f, reg[15](a5   ) = 0x0000000000000000
reg[16](a6   ) = 0x0000000000000200, reg[17](a7   ) = 0x0000000000000004
reg[18](s2   ) = 0x0000000000000200, reg[19](s3   ) = 0x0000000000000200
reg[20](s4   ) = 0x0000000000000001, reg[21](s5   ) = 0x00000000801dc89f
reg[22](s6   ) = 0x0000000000000000, reg[23](s7   ) = 0x0000000000000001
reg[24](s8   ) = 0x0000000000000000, reg[25](s9   ) = 0x00000000000001ff
reg[26](s10  ) = 0x0000000000000001, reg[27](s11  ) = 0xffffffffffffffff
reg[28](t3   ) = 0x0000000000000020, reg[29](t4   ) = 0x0000000000000002
reg[30](t5   ) = 0x0000000000000004, reg[31](t6   ) = 0x0000000000000000
freg[00](ft0 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e468()
freg[02](ft2 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e478()
freg[04](ft4 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e488()
freg[06](ft6 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e498()
freg[08](fs0 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e4c0()
freg[10](fa0 ) = 0x000000003b8efb12(), freg[-2147483648]() = 0x000000008017e4f0()
freg[12](fa2 ) = 0x0000000043700000(), freg[00]() = 0x000000008017e518()
freg[14](fa4 ) = 0x0000000047a8c000(), freg[00]() = 0x000000008017e528()
freg[16](fa6 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e538()
freg[18](fs2 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e560()
freg[20](fs4 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e570()
freg[22](fs6 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e580()
freg[24](fs8 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e590()
freg[26](fs10) = 0x0000000000000000(), freg[00]() = 0x000000008017e5a0()
freg[28](ft8 ) = 0x0000000000000000(), freg[00]() = 0x000000008017e5c8()
freg[30](ft10) = 0x0000000000000000(), freg[00]() = 0x000000008017e5d8()
W (158242305655) SYSCALL: sys_exit called by core 0 with 0x539
```

I used `riscv64-unknown-elf-objdump -lSD maixpy.elf > output.S` to get the line information where the crash occurs.

```
/home/.../MaixPy/ports/k210-freertos/platform/sdk/kendryte-standalone-sdk/lib/drivers/spi.c:380 (discriminator 3)
                buf[cmd_len / 4 + i] = ((uint32_t *)tx_buff)[i];
    800afc3e:   00fa8733                add     a4,s5,a5
    800afc42:   4318                    lw      a4,0(a4)
    800afc44:   009786b3                add     a3,a5,s1
    800afc48:   96a2                    add     a3,a3,s0
    800afc4a:   c298                    sw      a4,0(a3)
    800afc4c:   0791                    addi    a5,a5,4
    800afc4e:   b775                    j       800afbfa <spi_send_data_standard_dma+0xca>

```

I just figured out what crashes, but do not know why or what the cause is.

Any ideas?

Best wishes!

Vinschni, 2019-07-03:
Initializing the UART once and using it constantly, ensuring the RX buffer never overflows allows stable usage. With usage of `UART.deinit()` and reinitialization if needed I do not see stable results.

Neutree, 2020-08-08:
it seems caused by the convert(uint8_t* to uint32_t*), maybe the address of tx_buff not 4Bytes align