sipeed/MaixPy-v1 #115: Build Error: NATIVE_COMMAND undefined

robert-hh, 2019-07-06:
Building according to build.md runs into an error:
```
CMake Error at /home/robert/Downloads/MicroPython/MaixPy/components/micropython/CMakeLists.txt:305 (separate_arguments):
  separate_arguments given unknown argument NATIVE_COMMAND

CMake Error at /home/robert/Downloads/MicroPython/MaixPy/components/micropython/CMakeLists.txt:306 (separate_arguments):
  separate_arguments given unknown argument NATIVE_COMMAND

CMake Error at /home/robert/Downloads/MicroPython/MaixPy/components/micropython/CMakeLists.txt:383 (separate_arguments):
  separate_arguments given unknown argument NATIVE_COMMAND

CMake Error at /home/robert/Downloads/MicroPython/MaixPy/components/micropython/CMakeLists.txt:384 (separate_arguments):
  separate_arguments given unknown argument NATIVE_COMMAND
```

Neutree, 2019-07-06:
what OS and cmake you use? I just compiled on ubuntu18.04 with `cmake v3.10.2`

robert-hh, 2019-07-06:
debian 9.0,  cmake version 3.7.2

Neutree, 2019-07-06:
ummm, cmake 3.7.2 not support this command: https://cmake.org/cmake/help/v3.7/command/separate_arguments.html?highlight=separate_arguments

Neutree, 2019-07-06:
Add this command is for windows in the future, so maybe it's better to upgrade cmake to v3.9.6 or later

robert-hh, 2019-07-06:
I reinsatlled cmake and the toolchain, and then it worked.

closed 2019-07-06 by robert-hh