sipeed/MaixPy-v1 #116: Rebuilding the FS

robert-hh, 2019-07-06:
After I have now managed to rebuild the images, I have two new topics:
1. How do I rebuild the file system on the device. The existing FS does not allow to read anything
2. How do I include other python scripts in the image. Just copying them to .../MaixPy/components/micropython/port/builtin_py an rebuilding is not sufficient.

Neutree, 2019-07-06:
just copy them to `components/micropython/port/builtin_py` is right, and execute `python3 project.py rebuild`

Neutree, 2019-07-06:
The makefiles is generated from cmake int the new construction system, so do not edit the makefiles manually.
And you have to format your file system by `os.flash_format()` first for some reason (file path change in source maixpy code)

robert-hh, 2019-07-06:
Thanks. I found flash_format().
Still there is a hiccup: If I change a file in components/micropython/port/builtin_py, `python3 project.py rebuild` will not rebuild that changed file. I have to use clean and build.

robert-hh, 2019-07-17:
Hiccup gone. Thanks.

closed 2019-07-17 by robert-hh