sipeed/MaixPy-v1 #121: MICROPY_PY_THREAD (1) build issue

olahBalazs, 2019-07-14:
I have a build script (attached if you are interested
[build_script.zip](https://github.com/sipeed/MaixPy/files/3390421/build_script.zip)
).

I tried to enable the multi threading. I edited the file MaixPy/components/micropython/port/include/mpconfigport.h like:

```
#define MICROPY_PY_THREAD                   (1)
#define MICROPY_PY_THREAD_GIL               (1)
```

Then when I tried to build I got the following error:
[image]
Error: Implicit declaration of function  'free'

The error location is: MaixPy/components/micropython/port/src/mpthreadport.c line 190 column 9.

I then edited the file: MaixPy/components/micropython/port/src/mpthreadport.c  added
`#include <stdlib.h>`
at the top. Then the build successfully finished. Would it be possible to add this fix to the repo? Does anybody else have this issue?

By making a self build I lost the cpufreq python module. Does anybody know where I can re-enable it?

Neutree, 2019-07-20:
The Freertos from Kendryte have some problems, so we disabled it currently

Neutree, 2020-08-07:
enable it in menuconfig `python project.py menuconfig` now

closed 2020-08-07 by Neutree