sipeed/MaixPy-v1 #122: Sensor reset fails with different OV7725 module

Vinschni, 2019-07-17:
We want to utilize  HDF25-A50 OV7725 camera modules (like [those](https://wholesaler.alibaba.com/product-detail/30W-HD-Module-HDF25-A50-OV7725_60379832547.html)). As they are OV7725 modules as supported by MaixPy  we expected them to work.

We get no camera image and following error:
```error gc0328 detect, ret id is 0xff
[MAIXPY]: sensor reset function is null

Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
OSError: Reset Failed```

Any guess why those modules do not work?

Best regards!

Neutree, 2019-07-18:
not support 7725

closed 2019-07-18 by Neutree

Vinschni, 2019-07-18:
Which sensors are supported?

I could add support for OV7725, do you see any principal blocker for that?

Neutree, 2019-07-18:
currently ov2640 and ov7740, the voltage supply on board may not satisfy ov7725's requirement

jumpwatts, 2022-04-01:
@Neutree What voltage modification should be done to the board ? Should the DVDD be set to 1.8V?