sipeed/MaixPy-v1 #123: master build fail

thepian, 2019-07-19:
I'm trying to build a MaixPy binary with the new approach. I use a docker image that worked fine with the previous build structure.

1) I'm not quite sure of the correct order of commands for a first time build
2) clean_conf: flash.py is somehow needed
3) build: it bombs on a source repository

https://github.com/thepia/TalkiPy/tree/feature/docker

> docker build --tag k210 .
> docker run -i -v /Volumes/Projects/TalkiPy:/TalkiPy -t k210:latest /bin/bash

```
micropython@8704be20755f:/$ cd TalkiPy/projects/maixpy_k210
micropython@8704be20755f:/TalkiPy/projects/maixpy_k210$ python3 project.py --toolchain /opt/kendryte-toolchain/bin --toolchain-prefix riscv64-unknown-elf- config
-- SDK_PATH:/TalkiPy
maixpy
generate config file at: .config.mk
config complete
micropython@8704be20755f:/TalkiPy/projects/maixpy_k210$ python3 project.py clean_conf
-- SDK_PATH:/TalkiPy
maixpy
clean now
call flash.py error
micropython@8704be20755f:/TalkiPy/projects/maixpy_k210$ python3 project.py clean_conf
-- SDK_PATH:/TalkiPy
maixpy
clean now
call flash.py error
micropython@8704be20755f:/TalkiPy/projects/maixpy_k210$ python3 project.py conf
-- SDK_PATH:/TalkiPy
maixpy
usage: project.py [-h] [-p PORT] [-b BAUDRATE] [-t] [-n] [-s]
                  [-B {dan,bit,bit_mic,goE,goD,maixduino,kd233,auto}] [-S]
                  [--toolchain PATH] [--toolchain-prefix PREFIX] [--verbose]
                  {config,build,rebuild,menuconfig,clean,distclean,clean_conf,flash}
project.py: error: argument cmd: invalid choice: 'conf' (choose from 'config', 'build', 'rebuild', 'menuconfig', 'clean', 'distclean', 'clean_conf', 'flash')
micropython@8704be20755f:/TalkiPy/projects/maixpy_k210$ python3 project.py config
-- SDK_PATH:/TalkiPy
maixpy
config complete
micropython@8704be20755f:/TalkiPy/projects/maixpy_k210$ python3 project.py build
-- SDK_PATH:/TalkiPy
maixpy
build now
CMake Error: The source directory "" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [Makefile:372: cmake_check_build_system] Error 1
micropython@8704be20755f:/TalkiPy/projects/maixpy_k210$ 

```

thepian, 2019-07-19:
Additionally the spiffs checkout also seems to be missing.

```
Cloning into '/Volumes/Projects/TalkiPy/components/micropython/core/lib/lwip'...
Submodule path 'components/micropython/core/lib/axtls': checked out '43a6e6bd3bbc03dc501e16b89fba0ef042ed3ea0'
Submodule path 'components/micropython/core/lib/berkeley-db-1.xx': checked out '35aaec4418ad78628a3b935885dd189d41ce779b'
Submodule path 'components/micropython/core/lib/libffi': checked out 'e9de7e35f2339598b16cbb375f9992643ed81209'
Submodule path 'components/micropython/core/lib/lwip': checked out '92f23d6ca0971a32f2085b9480e738d34174417b'
Submodule path 'components/micropython/core/lib/mbedtls': checked out '3f8d78411a26e833db18d9fbde0e2f0baeda87f0'
Submodule 'crypto' (https://github.com/ARMmbed/mbed-crypto) registered for path 'components/micropython/core/lib/mbedtls/crypto'
Cloning into '/Volumes/Projects/TalkiPy/components/micropython/core/lib/mbedtls/crypto'...
Submodule path 'components/micropython/core/lib/mbedtls/crypto': checked out 'a78c958b17d75ddf63d8dd17255b6379dcbf259f'
Submodule path 'components/micropython/core/lib/nrfx': checked out '7513fc9d5c10dc28b25bf2bb6ff0ba66cb2a3c3d'
Submodule path 'components/micropython/core/lib/stm32lib': checked out '615329f1a61bd0689bfb095cb3fd74865cca88ff'
fatal: could not get a repository handle for submodule 'tools/spiffs/mkspiffs'
fatal: not a git repository: /Volumes/Projects/TalkiPy/tools/spiffs/mkspiffs/../../../.git/modules/tools/spiffs/mkspiffs
Unable to find current revision in submodule path 'tools/spiffs/mkspiffs'
```

Neutree, 2019-07-20:
you can refer to travis build script: https://github.com/sipeed/MaixPy/blob/master/.travis.yml

Neutree, 2019-07-20:
And build log here: https://travis-ci.org/sipeed/MaixPy

Neutree, 2019-07-20:
I wrote one dockerfile here: https://github.com/sipeed/MaixPy/blob/master/tools/docker/Dockerfile

thepian, 2019-07-23:
Somehow building works in your Docker. Thank you. The binary produced doesn't seem to work on the Dock I tried it on, and couldn't be uploaded to my Go.

chegewara, 2019-07-23:
@Neutree Thanks for this docker, it saved me a lot of time and let me to prepare this PR just in 1 day.

@thepian I am using maixduino board and it works great to me:
```
[MAIXPY]Pll0:freq:832000000
[MAIXPY]Pll1:freq:398666666
[MAIXPY]Pll2:freq:45066666
[MAIXPY]cpu:freq:416000000
[MAIXPY]kpu:freq:398666666
[MAIXPY]Flash:0xc8:0x17
open second core...
gc heap=0x8016f0e0-0x801ef0e0
[MaixPy] init end

 __  __              _____  __   __  _____   __     __
|  \/  |     /\     |_   _| \ \ / / |  __ \  \ \   / /
| \  / |    /  \      | |    \ V /  | |__) |  \ \_/ /
| |\/| |   / /\ \     | |     > <   |  ___/    \   /
| |  | |  / ____ \   _| |_   / . \  | |         | |
|_|  |_| /_/    \_\ |_____| /_/ \_\ |_|         |_|

Official Site : https://www.sipeed.com
Wiki          : https://maixpy.sipeed.com

MicroPython v0.4.0-15-gba05f8528-dirty on 2019-07-23; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>>
```

thepian, 2019-07-23:
I see.

1) seg fault with old boot.py on SD card
2) repl doesn't work. Typing has no effect. I see nothing unusual to give me hint.
3) I cannot connect via OpenML derived IDE

This goes for all of the master binaries that I tried.

Neutree, 2019-07-24:
@thepian  From your log, It shows program crashed in `wav_play_process`, so what you `boot.py` do?

closed 2020-08-07 by Neutree