sipeed/MaixPy-v1 #127: Maixduino board, SD and wifi possibly conflict

chegewara, 2019-07-24:
I am trying to test esp32 wifi API with this demo code, but is seems to crash on SD open file function:
```
import socket
import network
import gc
import os
import lcd, image
import machine
from board import board_info
import utime
from Maix import GPIO
from fpioa_manager import *

#iomap at MaixDuino
fm.register(25,fm.fpioa.GPIOHS10)#cs
fm.register(8,fm.fpioa.GPIOHS11)#rst
fm.register(9,fm.fpioa.GPIOHS12)#rdy
fm.register(28,fm.fpioa.GPIOHS13)#mosi
fm.register(26,fm.fpioa.GPIOHS14)#miso
fm.register(27,fm.fpioa.GPIOHS15)#sclk

nic = network.ESP32_SPI(cs=fm.fpioa.GPIOHS10,rst=fm.fpioa.GPIOHS11,rdy=fm.fpioa.GPIOHS12,
mosi=fm.fpioa.GPIOHS13,miso=fm.fpioa.GPIOHS14,sclk=fm.fpioa.GPIOHS15)

wifi = nic.scan()
for v in wifi:
    print("%s\n" %(v), end=" ")
print()

nic.isconnected()
nic.connect("sipeed", "12345678")
nic.isconnected()

sock = socket.socket()
addr = socket.getaddrinfo("dl.sipeed.com", 80)[0][-1]
sock.connect(addr)
sock.send('''GET / HTTP/1.1
Host: dl.sipeed.com
cache-control: no-cache

''')

utime.sleep_ms(300) # some delay is required here (most likely shorter), to let esp32 gatter some data, or esp32 will return data len 0
img = b""
sock.settimeout(10)
while True:
    data = sock.recv(512)
    if len(data) == 0:
        break
    print("rcv:", len(data))
    img = img + data

print(len(img))
sock.close()

img = img[img.find(b"\r\n\r\n")+4:]
print(len(img))
print("save to /sd/Alice.jpg")
f = open("/sd/Alice.jpg","wb")
f.write(img)
f.close()
print("save ok")
print("display")
img = image.Image("/sd/Alice.jpg")
lcd.init()
lcd.display(img)
```
With this error:
```
[MaixPy] parse_ipv4_addr | It is not string IP form
*** Get host by name
5011630241*** Get socket
Allocated socket #0
*** Socket connect mode:0
Conn Connection: WL_CONNECTED
64
ESPSocket: 2826 bytes available
Reading 2826 bytes from ESP socket with status SOCKET_ESTABLISHED
rcv: 2826
ESPSocket: 0 bytes available
2826
2578
save to /sd/Alice.jpg
Traceback (most recent call last):
  File "<stdin>", line 53, in <module>
OSError: [Errno 5] EIO
```

chegewara, 2019-08-03:
Ok, finally i found this conflict.
If not count that i am re-register function for pins 26-28, which i should not do, because those pins already have registered functions on system start, it is not possible to use SD CARD and esp32 wifi at the same time.

The reason is very simple. SPI pins that are common for SD CARD and esp32 spi are in conflict.
In sd card driver SPI pins are not gpiohs
https://github.com/sipeed/MaixPy/blob/master/components/drivers/sd_card/src/sdcard.c#L421-L424
which is mandatory for esp32 SPI driver:
https://github.com/sipeed/MaixPy/blob/master/components/micropython/port/src/standard_lib/network/esp32/esp32_spi_io.c#L29-L35

Neutree, 2019-08-03:
sd card and wifi use the same SPI, refer to schematic: http://dl.sipeed.com/MAIX/HDK/Maixduino/Maixduino-4.30/Maixduino-4.30%28schematic%29.pdf

chegewara, 2019-08-03:
Yes, i know they are using the same SPI.
Maybe i am doing something wrong, but SD card miso, mosi and sclk functions are registered with low speed GPIO functions. esp32 SPI pins are registered with `gpiohs_set_drive_mode`. 
When i am trying to register esp32 SPI with the same functions for miso, mosi and sclk like they are registered in sd driver i am getting this error:
`framework-kendryte-standalone-sdk\lib\drivers\gpiohs.c:36) pin < GPIOHS_MAX_PINNO`
From this error i can say that so we can use only `FUNC_GPIOHS0` thru `FUNC_GPIOHS31` to init esp32 SPI.

Here is my changed esp32 init code that is causing above error:
```
#define ESP32_CS        FUNC_GPIOHS25
#define ESP32_RST       FUNC_GPIOHS11
#define ESP32_RDY       FUNC_GPIOHS12
#define ESP32_MOSI      FUNC_SPI1_D0
#define ESP32_MISO      FUNC_SPI1_D1
#define ESP32_SCLK      FUNC_SPI1_SCLK

....
    /* INIT SPI */
    // #iomap at MaixDuino
    fpioa_set_function(25, ESP32_CS);
    fpioa_set_function(8, ESP32_RST);
    fpioa_set_function(9, ESP32_RDY);
    fpioa_set_function(28, ESP32_MOSI);
    fpioa_set_function(26, ESP32_MISO);
    fpioa_set_function(27, ESP32_SCLK);

    esp32_spi_config_io(ESP32_CS - FUNC_GPIOHS0, ESP32_RST, ESP32_RDY - FUNC_GPIOHS0,
                        ESP32_MOSI - FUNC_GPIOHS0, ESP32_MISO - FUNC_GPIOHS0, ESP32_SCLK - FUNC_GPIOHS0);

....
```

chegewara, 2019-08-03:
Exactly this error:
https://github.com/sipeed/MaixPy/blob/master/components/micropython/port/src/standard_lib/network/esp32/mod_esp32.c#L194-L226

chegewara, 2019-08-03:
@Neutree 
OK, i found solution, just i dont know if this the correct one, because seems not to be very elegant.
Here is working micropython code to download `Alice.jpg` and display on LCD:
```
#/**
 #* Simple demo to test network
 #*/
import socket
import network
import gc
import os
import lcd, image
import machine
from board import board_info
import utime
from Maix import GPIO
from fpioa_manager import *

#iomap at MaixDuino
fm.register(25,fm.fpioa.GPIOHS10)#cs
fm.register(8,fm.fpioa.GPIOHS11)#rst
fm.register(9,fm.fpioa.GPIOHS12)#rdy
fm.register(28,fm.fpioa.GPIOHS13)#mosi
fm.register(26,fm.fpioa.GPIOHS14)#miso
fm.register(27,fm.fpioa.GPIOHS15)#sclk

nic = network.ESP32_SPI(cs=fm.fpioa.GPIOHS10,rst=fm.fpioa.GPIOHS11,rdy=fm.fpioa.GPIOHS12,
mosi=fm.fpioa.GPIOHS13,miso=fm.fpioa.GPIOHS14,sclk=fm.fpioa.GPIOHS15)

wifi = nic.scan()
for v in wifi:
    print("%s\n" %(v), end=" ")
print()

nic.isconnected()
nic.connect("sipeed", "12345678")
nic.isconnected()

sock = socket.socket()
addr = socket.getaddrinfo("dl.sipeed.com", 80)[0][-1]
sock.connect(addr)
sock.send('GET /MAIX/MaixPy/assets/Alice.jpg HTTP/1.1\r\nHost: dl.sipeed.com\r\ncache-control: no-cache\r\n\r\n')

utime.sleep_ms(300) # some delay is required here (most likely shorter), to let esp32 gatter some data, or esp32 will return data len 0
img = b''
sock.settimeout(10)
while True:
    data = sock.recv(1412)
    if len(data) == 0:
        break
    print("rcv:", len(data))
    img = img + data

print(len(img))
img = img[img.find(b"\r\n\r\n")+4:]
print(len(img))
// here we need to re-register SPI common pins to match setting from sdcard.init()
fm.register(28,fm.fpioa.SPI1_D0)#mosi
fm.register(26,fm.fpioa.SPI1_D1)#miso
fm.register(27,fm.fpioa.SPI1_SCLK)#sclk
print("save to /sd/Alice.jpg")
f = open("/sd/Alice.jpg","wb")
f.write(img)
f.close()
print("save ok")
print("display")
img = image.Image("/sd/Alice.jpg")
lcd.init()
lcd.display(img)
```

ekthomson, 2019-08-07:
They don't exactly "conflict" but they're using the same SPI bus, SPI0. They have separate chip select lines which should allow them to both use the same bus alternately. The problem is that you're reassigning the pins for the SPI0 to high speed GPIO instead, which yanks them away from the SPI driver used by the SD card. I've not used them together, but theoretically if you can get the ESP driver to use the machine.SPI for its communications with the correct chip select, they should be able to be used alternately without having to toggle between GPIO and hardware SPI. 

At least that's the theory. I've found that using the SPI1 with machine.SPI seems to break SPI0 and SD card as well with release 3.2, so there's some other fundamental breakage in the machine.SPI code on the maixpy boards. I haven't tracked down what's causing it yet though, and even the chip select on 0.3.2 seems to be broken in machine.SPI as I get an error "ValueError: [MAIXPY]SPI: cs value error
" every time I try to write with the cs flag set.

chegewara, 2019-08-07:
@ekthomson I am reassign SPI pins, because i found it is only way to do it without changing SD card or esp32 driver code.

nogleo, 2020-07-28:
I've been having some problem with this  bus sharing between esp32 and sd slot on my Maixduino. I'm makin this project where a I need to collect data from sensors, at least one is through the ADC, and log them ins the sd card.  If I didn't need a high frequency it would be possible to just chip select from one to the other...  Is there a way to reassign the sd pins to unused digital ones so I can use an external sd adapter?

junhuanchen, 2020-11-12:
fix this https://github.com/sipeed/MaixPy/commit/41a81b2485fb57f244171e5f69d733735335c931

https://github.com/sipeed/MaixPy_scripts/blob/master/network/demo_https_get.py

closed 2020-11-12 by junhuanchen