sipeed/MaixPy-v1 #135: OV7740设置水平镜像图像变色

kingsjl, 2019-07-29:
不设置水平镜像的时候颜色是正常的
sensor.set_hmirror(0)

[image]

设置水平镜像颜色就不正常了
sensor.set_hmirror(1)
[image]

Neutree, 2019-08-07:
根据手册所述，设置ov7740的镜像翻转是设置寄存器0x0C的第6位为1，代码也是这样写的，但是却出现了颜色不对的情况

kingsjl, 2019-08-08:
> 根据手册所述，设置ov7740的镜像翻转是设置寄存器0x0C的第6位为1，代码也是这样写的，但是却出现了颜色不对的情况

这个意思是无解了吗

kingsjl, 2019-08-08:
https://github.com/raspberrypi/linux/blob/4a58d02765ba6c28776778fe75a85960b7658400/drivers/media/i2c/ov7740.c

> 根据手册所述，设置ov7740的镜像翻转是设置寄存器0x0C的第6位为1，代码也是这样写的，但是却出现了颜色不对的情况

https://github.com/raspberrypi/linux/blob/4a58d02765ba6c28776778fe75a85960b7658400/drivers/media/i2c/ov7740.c

Neutree, 2020-08-07:
暂时使用了软件翻转

closed 2020-08-07 by Neutree