sipeed/MaixPy-v1 #139: Master branch binaries do not work for me

phjo33, 2019-08-02:
On both my Maixduino and maix-bit boards, I flashed maixpy_v0.4.0_30_g0fc8098 and maixpy_v0.4.0_17_g1212f5d firmwares. Supposedly, the flashing was successfull, but nothing on the screen, and impossible to connect to the device with maixpy ide or rshell. I tried to reboot each device to no avail.

No such problem with the 0.3.2 release. The "Welcome to MaixPy" is soon displayed on the screen, and I can connect to the device (maix-bit or maixduino) easily.

I must be doing something stupid, but I have no idea what !

Neutree, 2019-08-02:
0.4.0 have changed fs(file system), so you have to format your fs with `os.flash_format()` or upload [this package](http://dl.sipeed.com/MAIX/MaixPy/release/erase_spiffs.kfpkg) to flash(board) to erase fs and the firmware will rebuild the fs on the startup.

Neutree, 2019-08-02:
and if you use IDE, you should upgrade to 0.2.3:http://dl.sipeed.com/MAIX/MaixPy/ide/v0.2.3/

phjo33, 2019-08-02:
It worked, thanks !

closed 2019-08-02 by phjo33