sipeed/MaixPy-v1 #142: Latest MaixPy v.0.4.0_30 with 20class objects

amriddell, 2019-08-03:
I have 20 class objects example working with MaixPy v0.3.2 but using the same firmware approach for v0.4.0_30 through kflash_gui.app doesn't work.

Am I missing something here?

Thanks,
Alistair

Vinschni, 2019-08-09:
Did you ensure to flash the model to the same address in flash you do `kpu.load()` from? 
Did you ensure not to overwrite your 20class model in flash with a part of the firmware .bin?

amriddell, 2019-08-10:
So I produced:

> zip maixpy_v0.4.0_30_min.bin.kfpkg maixpy_v0.4.0_30_min.bin 20class.kmodel flash-list.json

The flash-list.json is:

{
  "version": "0.1.1",
  "files": [
    {
      "address": 0x000000,
      "bin": "maixpy_v0.4.0_30_min.bin",
      "sha256Prefix": true
    },
    {
      "address": 0x500000,
      "bin": "20class.kmodel",
      "sha256Prefix": false
    }
  ]
}

In the FaceShow.py example, I use this:

classes = ['aeroplane', 'bicycle', 'bird', 'boat', 'bottle', 'bus', 'car', 'cat', 'chair', 'cow', 'diningtable', 'dog', 'horse', 'motorbike', 'Person', 'pottedplant', 'sheep', 'sofa', 'train', 'tvmonitor']
task = kpu.load(0x500000)
anchor = (1.889, 2.5245, 2.9465, 3.94056, 3.99987, 5.3658, 5.155437, 6.92275, 6.718375, 9.01025)
kpu.init_yolo2(task, 0.5, 0.3, 5, anchor)

The script seems to get stuff and nothing appears on the display.

Is there something wrong with the above info?

Thanks for your response.

Alistair

Vinschni, 2019-08-10:
Does an error occur on repl/debug output?

amriddell, 2019-08-10:
The new version of MaixPy seems to work OK.
It’s the 20 class addition that I don’t think is installed properly.

I have not seen any errors probably because I upload py files through pyLoader

I configure it again and run the real from the command line and see what that says.

amriddell, 2019-08-10:
I just installed the current MaixPy version again (v.0.4.0_30) with the 20class and it is working fine now.
I think I thought it wasn’t working because it took a while to start up.

I’m running the binocular camera hardware with two wide angle cameras installed.
It is actually running better that it was before.

Thanks for your response and interest. 

This is my test code:
I’m not sure I need the sensor.binocular_reset() call. I’ll test it without it later.

#
# Id a class from a choice of 20 and flash Green LED.
# Continue to modify this with new functionality.
# Uses Class and likelihood number in frame
#
# DLM:  Sun Jul  7 16:24:16 AEST 2019
#
from board import board_info
from hwconfig import LED
import utime, sensor, image, lcd
import KPU as kpu

fm.register(board_info.LED_R, fm.fpioa.GPIO0)
#fm.register(board_info.LED_G, fm.fpioa.GPIO4)
fm.register(board_info.PIN15, fm.fpioa.GPIO5) # Centre Push Button for exit/reset
led_r = GPIO(GPIO.GPIO0, GPIO.OUT)
#led_g = GPIO(GPIO.GPIO4, GPIO.OUT)
input = GPIO(GPIO.GPIO5, GPIO.IN)

lcd.init()
lcd.freq(15000000)

sensor.reset()
sensor.binocular_reset()

# Single camera only here
sensor.shutdown(False)
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.set_hmirror(0) # flip camera
sensor.set_vflip(1)

# Dual if uncommented and added in
sensor.shutdown(True)
sensor.set_pixformat(sensor.RGB565)
sensor.set_framesize(sensor.QVGA)
sensor.set_vflip(1)

sensor.run(1)
lcd.clear()

classes = ['aeroplane', 'bicycle', 'bird', 'boat', 'bottle', 'bus', 'car', 'cat', 'chair', 'cow', 'diningtable', 'dog', 'horse', 'motorbike', 'Person', 'pottedplant', 'sheep', 'sofa', 'train', 'tvmonitor']
task = kpu.load(0x500000) # This varies and normally is 0x300000
anchor = (1.889, 2.5245, 2.9465, 3.94056, 3.99987, 5.3658, 5.155437, 6.92275, 6.718375, 9.01025)
kpu.init_yolo2(task, 0.5, 0.3, 5, anchor)

#utime.localtime(utime.time())

while(True):
#    clock.tick()
    if input.value() == 0:
        led_r.value(1)
#        led_b.value(1)
        break
    led_r.value(1)
#    led_r.value(0)
    img = sensor.snapshot()
    code = kpu.run_yolo2(task, img)
    if code:
        led_r.value(0)
#        led_r.value(1)
        for i in code:
            img.draw_rectangle(i.rect())
            lcd.display(img)
            for i in code:
                lcd.draw_string(i.x(), i.y(), (classes[i.classid()] + ' ' +'%f'%i.value()) , lcd.RED, lcd.WHITE)
#                lcd.draw_string(i.x(), i.y()+12, '%f1.3'%i.value(), lcd.RED, lcd.WHITE)
    else:
        lcd.display(img)
kpu.deinit(task)
#led_g.value(1)
led_r.value(1)
fm.unregister(board_info.LED_R, fm.fpioa.GPIO0)
#fm.unregister(board_info.LED_G, fm.fpioa.GPIO4)
fm.unregister(board_info.PIN15, fm.fpioa.GPIO5)

print("Find Face 3 Finished.\r\r")

sensor.shutdown(True)
sensor.reset()
sensor.binocular_reset()
utime.sleep_ms(10)
machine.reset()

Vinschni, 2019-09-20:
Please close the issue as the problem seems fixed.

closed 2019-09-26 by amriddell