sipeed/MaixPy-v1 #145: OV2640 sensor.reset() error

Vinschni, 2019-08-08:
I brought [those OV2640 modules](https://www.banggood.com/5pcs-Lengthen-Fisheye-Lens-TTGO-Camera-Module-OV2640-2-Megapixel-Adapter-Support-YUV-RGB-JPEG-For-T-Camera-Plus-ESP32-DOWDQ6-8MB-SPRAM-p-1493567.html?) after [i learned](https://github.com/sipeed/MaixPy/issues/122) that the OV7725 are probably not compatible.

Now with those OV2640 and a new MaixBit board i get the same error as before:
We get no camera image and following error on `sensor.reset()`
```
error gc0328 detect, ret id is 0xff
[MAIXPY]: sensor reset function is null

Traceback (most recent call last):
  File "<stdin>", line 12, in <module>
OSError: Reset Failed
```

Any hints what the problem could be?

Vinschni, 2019-08-08:
I compiled the latest [commit](https://github.com/sipeed/MaixPy/commit/b29aa57e6abc899ddc2b55629757897c57abeebe) . With that commit it works.

closed 2019-08-08 by Vinschni

MengXiao92, 2019-12-31:
I met the same problem with OV2640 and a new MAiX GO board, can you tell me how deal with?

Vinschni, 2020-01-03:
I solved the problem ensuring to at least using https://github.com/sipeed/MaixPy/commit/b29aa57e6abc899ddc2b55629757897c57abeebe

bwellmeshed, 2020-01-13:
Hello I am having the same issue as well what do you mean by at least using b29aa57

Vinschni, 2020-01-13:
I recommend using / [building](https://github.com/sipeed/MaixPy/blob/master/build.md) at least MaixPy with commit b29aa57

bwellmeshed, 2020-01-13:
Okay thanks

JrHdez, 2020-06-17:
> Okay thanks

Hi! I'm new at this, and I also have the same problem: When executin the next line of code: sensor.reset(), I got:

init i2c2
[mt9dll] Sensor ID:0xffff,req:0x1519
error mt9dlll detect, ret id is 0xffff
[MAIXPY]: no sensor
Traceback (most recent call last):
 File "<stdin>", line1, in <module>
OSError: Reset Failed

Pls help! Thanks!