sipeed/MaixPy-v1 #146: No response to keyboard in serial terminal connection to MaixPy Go

pscheie, 2019-08-09:
(This is an updated message I posted incorrectly in the MaixPy Scripts project which is the wrong place for this problem.)

I can connect to my MaixPy via serial cable and I get the MaixPy welcome screen, but it doesn't respond at all to my keystrokes. I'm running minicom on Linux Mint connected to /dev/ttyACM0 at 115200 with hardware and software flow control set to OFF, and the backspace key set to DEL, and Line wrap set to YES. I've tried the USB cable that came with the MaixPy and one of my own USB cables. I also tried using gtkterm instead of minicom. They all do the same thing: I get the MaixPy greeting screen and the MicroPython >>> prompt, but when I hit Enter or Ctrl-C or any other key nothing happens.

Here's what minicom looks like when I connect:

Welcome to minicom 2.7.1

OPTIONS: I18n
Compiled on Aug 13 2017, 15:25:34.
Port /dev/ttyACM0, 20:57:40

Press CTRL-A Z for help on special keys

| \ / | /\ |_ | \ \ / / | _ \ \ \ / /
| \ / | / \ | | \ V / | |) | \ _/ /
| | / | / /\ \ | | > < | / \ /
| | | | / __ \ | | / . \ | | | |
| _ | | _ | / / \ _\ |_| / / \ \ || |_|
Official Site:http://www.sipeed.com/
Wiki:http://maixpy.sipeed.com/
[MAIXPY]Pll0:freq:320666666
[MAIXPY]Pll1:freq:159714285
[MAIXPY]Flash:0xc8:0x17
[MAIXPY]:Spiffs Mount successful
[MAIXPY]LCD:init

MicroPython 3b5c45649-dirty on 2018-11-15; Sipeed_M1 with kendryte-k210
Type "help()" for more information.

CTRL-A Z for help | 115200 8N1 | NOR | Minicom 2.7.1 | VT102 | Offline | ttyACM0

Any idea how to fix this? Is something on my MaixPy not set correctly? Minicom over serial works fine with other boards such as my BeagleBone Black (using an FTDI USB-to-serial adapter). Thanks.

pscheie, 2019-08-09:
Update: I forgot to mention that I was getting the white-on-blue "Welcome to MaixPy" screen when I first power it, but that it went black/dark when I connected to it with minicom.

I found that the kflash_gui was able to talk to the MaixPy, so I uploaded maixpy_v0.3.2_full.elf to it, but now I don't get anything on the LCD screen even when I power it on, and when I try to connect with minicom I no longer get the MaixPy welcome screen.  And I still get no response to any keystrokes in minicom.  So, now the MaixPy appears to be dead, although I think I can still upload images with kflash_gui.  Any help?

robert-hh, 2019-08-11:
Which board are using? Being able to upload the firmware indicates proper communication in both directions.

pscheie, 2019-08-11:
I've got the MaixPy Go.  Yes, I agree that the fact that the kflash_gui works is a good sign that the board itself is working, although as I said, nothing comes up on the screen anymore after I uploaded maixpy_v0.3.2_full.elf to it.  Should i try uploading another firmware image to it?

robert-hh, 2019-08-11:
I would try another terminal emulator first. After a few attempts, I use picocom now on Linux. small & easy.

pscheie, 2019-08-11:
Using picocom, I'm seeing this:

port is        : /dev/ttyACM0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Type [C-a] [C-h] to see available commands

Terminal ready
sting, something's wrong, boot failed with exit code 233, go to find your vendor.

I think that last line is a problem.

pscheie, 2019-08-11:
Using kflash_gui, I uploaded maixpy_v0.3.2_minimum.elf to the maixpy.  kflash_gui reported that the upload (it said download) was successful.  I cycled the power on the maixpy and tried to connect to it serialy  using picocom, but I'm still getting the same error:

"interesting, something's wrong, boot failed with exit code 233, go to find your vendor."

Noting that the first word is not capitalized, I don't think it's the first word in the error, but meaning is clear.  Actually, the first time I connected with picocom, the word "interesting" was missing, and the messages started at the comma.  So, I disconnected, reconnected, and then the message appeared starting with "esting, something's wrong...".  I disconnected, reconnected, and then I got the whole word "interesting" at the beginning of the message.  I disconnected and reconnected a couple more times, but the message never started with any words prior to 'interesting'.

robert-hh, 2019-08-12:
Looking at pictures and data sheets of the Seed Studio Maix Go board, they seem to use a STM32F103C8 (or similar) controller for the USB/UART bridge. So the error message could be generated by that chip too.

Neutree, 2020-08-07:
... @pscheie upload error firmware.. not .elf file, should be .bin file

closed 2020-08-07 by Neutree