sipeed/MaixPy-v1 #148: esp32 firmware erased

vagoverso, 2019-08-13:
I messed up the esp32 in my maixduino. How can I find the firmware?

Neutree, 2019-08-14:
http://dl.sipeed.com/MAIX/HDK/factory_firmware/esp32/

chegewara, 2019-08-14:
@Neutree Nice, it is finally available, thanks. I have question about esp32 firmware, i made some tweaks in sdkconfig and i think my maixduino works better with downloading files (not perfect yet), are you or whoever is responsible to cooperate?

Neutree, 2019-08-14:
@xelll

vagoverso, 2019-08-14:
Thank you very much. I used esptool to flash it. First I've erased the flash then installed the bin on position 0x1000 but it didn' work (the led was blinking)- So I installed an official firmware from Espressif  an I've put the NINA bin at 0x100000 hoping the problem was the partition table or boot. I'm sorry I'm such a noob. If I try to run the example  MaixPy_scripts/network/demo_esp32_read_adc.py (the only one that uses esp32) I get this errors:

> 1
1
1
1
1
1
Reset ESP32
ESP32_SPI init over
ESP32 timed out on SPI select
ESP32 timed out on SPI select
esp32_spi_sample_adc: get resp error!
esp32_spi_get_adc_val: esp32_spi_sample_adc error!

Traceback (most recent call last):
  File "<stdin>", line 23, in <module>
ValueError: [MaixPy]: esp32 read adc failed!

MicroPython v0.4.0-31-gc1d8a43-dirty on 2019-08-06; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>> 
Hope you have some spare time,thanks.

chegewara, 2019-08-14:
@vagoverso You need to flash it from 0x00 address.

vagoverso, 2019-08-14:
@chegewara thanks. I did and now I have no blinking led but I have the same results. 
“According to the ESP IDF, the default pins for SPI(1) are: sck=Pin(6), mosi=Pin(8), miso=Pin(7). For SPI(2) it's: sck=Pin(14), mosi=Pin(13), miso=Pin(12). In MicroPython the CS pin must be handled by user code, it's not controlled by the hardware peripheral.” . Can PIN numbers of the example be wrong?

chegewara, 2019-08-14:
That strange, i didnt test fw from that link, but with original fw i could read ADC.

The io map you are using is correct. I have original fw v1.2.2, if i remember, so i will upload here later.

vagoverso, 2019-08-15:
Thanks! It would be great. Looking forward.

chegewara, 2019-08-15:
[maixduino_esp32_original.bin.txt](https://github.com/sipeed/MaixPy/files/3504477/maixduino_esp32_original.bin.txt)
Just rename and flash it.

vagoverso, 2019-08-15:
@chegewara you made my day! Everything works! Thank you very much. I felt like I had a Ferrari but no oil :)
First lesson learned: don't mess up with things you don't know. Ask someone first. That's what communities are for!
Now it's time to learn more. Thank you and thank you @Neutree .

chegewara, 2019-08-15:
@Neutree I can confirm that http://dl.sipeed.com/MAIX/HDK/factory_firmware/esp32/NINA_W122_33.bin cant read ADC.

chegewara, 2019-08-16:
Hi @Neutree @xelll 
I am playing a bit with NINA and MaixPy firmware and i found that only 5 ADC pins are read. Pin ADC0 on maixduino board is not read and sent with NINA firmware installed on maixduino board.

I can fix it easy and create PR but this require changes in both, esp32 NINA firmware and MaixPy esp32_spi driver.

nnn112358, 2019-08-18:
Hi @chegewara , @Neutree
I had the same trouble. Very helpful.

I tried fixed_adc_fw.tar.gz. It does not work. 
I tried am maixduino_esp32_original.bin.txt.  It work, good.

chegewara, 2019-08-18:
Hi @nnn112358 
yes, in fixed_adc_fw.tar.gz is FW that its flashed from regular make flash which should be flashed from 0x10000, my bad, sorry.
Here is full firmware that needs to be flashed from 0x0 address:
[NINA_W102-1.3.1.bin.txt](https://github.com/sipeed/MaixPy/files/3513695/NINA_W102-1.3.1.bin.txt)

chegewara, 2019-08-19:
Here is firmware with disabled DEBUG logging:
[fixed_adc_fw.tar.gz](https://github.com/sipeed/MaixPy/files/3514606/fixed_adc_fw.tar.gz)

nnn112358, 2019-08-22:
@chegewara 
your fixed_adc_fw.tar.gz is not working properly in maxduino.
Read ADC is OK.
But ,file download example is NG. File size is Small (5kB). 

File downloads example
https://github.com/chegewara/sipeed-maixduino-test-apps/blob/master/platformIO/MaixPy/esp32-wifi.py

Your information is useful and helpful. Thanks.

chegewara, 2019-08-23:
@nnn112358 Thanks for info, i will update MaixPy and NINA code on github so you and/or anyone else could use it.

chegewara, 2019-08-25:
https://github.com/chegewara/MaixDuino_ESP32_SpiSlave/tree/maixduino
https://github.com/chegewara/MaixPy/tree/adc_update

Both updated to work with all 6 ADC pins, not compatible with old firmware.

Neutree, 2020-08-07:
and latest ESP32 firmware can be find here: https://github.com/sipeed/Maixduino_esp32_fimware/releases

closed 2020-08-07 by Neutree

AndresFelipeGomez, 2021-01-10:
hi, I want to create a server with the maixduino, I am using the firmware maixpy_v0.6.2_6_gd78d036a2.bin that loads with kflash

import network
import utime

from Maix import GPIO
from fpioa_manager import *

fm.register(25,fm.fpioa.GPIOHS10)#cs
fm.register(8,fm.fpioa.GPIOHS11)#rst
fm.register(9,fm.fpioa.GPIOHS12)#rdy
fm.register(28,fm.fpioa.GPIOHS13)#mosi
fm.register(26,fm.fpioa.GPIOHS14)#miso
fm.register(27,fm.fpioa.GPIOHS15)#sclk

nic = network.ESP32_SPI(cs=fm.fpioa.GPIOHS10,rst=fm.fpioa.GPIOHS11,rdy=fm.fpioa.GPIOHS12,
mosi=fm.fpioa.GPIOHS13,miso=fm.fpioa.GPIOHS14,sclk=fm.fpioa.GPIOHS15)

if not nic.isconnected():
    nic.connect("GOMEZ", "1102381153")
    print("waitinig for connection...")
    while not nic.isconnected():
        utime.sleep(1)
print(nic.ifconfig())

import usocket as socket
addr = socket.getaddrinfo("dl.sipeed.com", 80)[0][-1]
s = socket.socket()
s.connect(addr)
s.send('''GET /MAIX/MaixPy/assets/Alice.jpg HTTP/1.1\r\nHost: dl.sipeed.com\r\ncache-control: no-cache\r\n\r\n''')
utime.sleep_ms(5000) # some delay is required here (most likely shorter), to let esp32 gatter some data, or esp32 will return data len 0
img = b""
s.settimeout(20)
while True:
    data = s.recv(1000)
    if len(data) == 0:
        break
    print("rcv:", len(data))
    img = img + data

print(len(img))
s.close()
nic.ifconfig()

buuuuutttt: OSError: [Errno 5] EIO