sipeed/MaixPy-v1 #149: master branch is broken

chegewara, 2019-08-16:
Here is what i have after building and flashing master branch:
```
[MAIXPY]Pll0:freq:832000000
[MAIXPY]Pll1:freq:398666666
[MAIXPY]Pll2:freq:45066666
[MAIXPY]cpu:freq:416000000
[MAIXPY]kpu:freq:398666666
[MAIXPY]Flash:0xc8:0x17
open second core...
gc heap=0x801700e0-0x801f00e0
[MaixPy] init end
Traceback (most recent call last):
  File "_boot.py", line 17, in <module>
NameError: name 'time' isn't defined
MicroPython v0.4.0-39-g083e0cca5-dirty on 2019-08-16; Sipeed_M1 with kendryte-k210
Type "help()" for more information.
>>> 
__class__       __name__        os              sys
devices         i
>>> 
```
If i build and flash from my repository, which is about 2 week behind, it works.

chegewara, 2019-08-16:
Looks like this commit did this: https://github.com/sipeed/MaixPy/commit/8394fd73aa02c251199b9843a00c50aeb45ea163
Reverting _boot.py solved problem.

Neutree, 2019-08-16:
fixed: https://github.com/sipeed/MaixPy/commit/95f00f0cabdcddbeac3fd6562dc8af481f7eaf6b

closed 2019-08-16 by Neutree

chegewara, 2019-08-16:
Thanks @Neutree