sipeed/MaixPy-v1 #155: kpu: load error -2/-3

gundamMC, 2019-09-06:
kpu.load throws error -2 when loading kmodel from flash (downloaded using kflash_gui) and -3 when loading from sd card.

What do the error codes mean?

And here's the output of kpu.memtest if it helps:  (My .kmodel file is 100KB)
```
###free gc heap memory : 490 KB
###free sys heap memory: 1360 KB
None
```

Code to replicate the issue:
```python
task = kpu.load(0x300000)
# task = kpu.load('/sd/out.kmodel')
```

Here's also a zip file containing the kmodel file on the sd card and the kfpkg file (which also contains the kmodel).
 [kmodel.zip](https://github.com/sipeed/MaixPy/files/3584873/kmodel.zip)

gundamMC, 2019-09-07:
It turns out that the problem is due to the version of nncase. When I use v 0.1.0 rc5, the error disappears. Nevertheless, I'm still kinda curious what the error codes mean (and why they were different when loading from flash and sd card).

Neutree, 2020-08-07:
it will print string error msg in latest firmware

v4 convert by nncase 2.0 beta 2 is supported

closed 2020-08-07 by Neutree