sipeed/MaixPy-v1 #160: About mpy

crouse12, 2019-09-14:
Is it possible to make mpy file for maix bit?

robert-hh, 2019-09-14:
Yes. No problem with mpy-cross. Th eproblem is getting a compiled version of mpy-cross. If you have set up your toolbox to compile the firmware youself, you will find it at:
.../MaixPy/components/micropython/core/mpy-cross
then you can compile your .py files e.g. with
`mpy-cross myfile.py`
which creates myfile.mpy. On the device, mpy files can be imported just like .py files.

crouse12, 2019-09-22:
I try to compile maixpy firmata. First, I compile my .py file to .mpy and upload to my maixbit.
However, the compiled .mpy file cannot work. I think this is because my firmata verion in maixbit is only v0.3.2. Therefore, I try to compile and upload maixpy.bin to my maixbit with
kflash maixpy.bin. maixpybin.py is in /MaixPy/maixpy_k210. However, my mpy file still cannot work.
Finally, I upload maixpy_v0.3.2_full.bin to my maixbit.  Unfortunately, boot.py cannot work now. I found that there are two boot.py and two freq.conf on my maixbit.  //boot.py cannot be deleted.

[image]

crouse12, 2019-09-22:
I found it, so maixbit works now. 
https://github.com/sipeed/MaixPy/issues/139
However, I still cannot compile my mpy file.
Could you give me v0.3.2 micropython firmata source code?

robert-hh, 2019-09-22:
which mpy-cross version did you use to compile the .py file. You have to take it from the build tree of the MaixPy firmware, not from a different source.

crouse12, 2019-09-22:
I download branch historic of maixpy and use it to compile my mpy.
This compiled mpy works now.

closed 2020-08-07 by Neutree

odudex, 2024-03-17:
Tried this now. Built mpy-cross within Maixpy micropython core, then used a boot.py file which call .mpy modules I compiled with mpy-cross.
The modules are found, but micropython accuses syntax error.
Is there any argument needed beyond `mpy-cross myfile.py`? I read that there's the `march` argument to specify architecture, but it seems RiscV is not supported.